iNavFlight/inav #272: Would like Angle Mode work up to 90 degree angle

DaleKramer, 2016-06-12:
For my application (see my last post to the below pull request), I need angle mode to work all the way up to a setting of 90 degrees (max_angle_inclination of 900). There is an unresolved pull request about this in CleanFlight [https://github.com/cleanflight/cleanflight/pull/1180](url).
Any help would be appreciated.
Dale

DzikuVx, 2016-06-12:
Never tried that, but according to code https://github.com/iNavFlight/inav/blob/master/src/main/io/serial_cli.c#L728 it is already done in INAV: max_angle_inclination_rll and max_angle_inclination_pit can go up to 900. 

@DaleKramer could you please confirm this is what you wanted and reopen if not?

DaleKramer, 2016-06-12:
From reading the Cleanflight issue, it seems to suggest that although you can ask for 900 you only get 500, so I still think this is likely still an issue.  I have not got a working ship to test this with.

DzikuVx, 2016-06-12:
INAV uses different PID controller than CF. Different code, different principles. You would have to test it first.
If works up to 90deg do nothing. If does not work, comment here, we will reopen

DzikuVx, 2016-06-12:
@DaleKramer OK, you are right, there is a bug and 500deg is max ANGLE can request.
So, reopened, tagged, sooner or later will be fixed

DaleKramer, 2016-06-12:
I am tracking this down because I don't want to have to test this on my $5,000 ship.

DzikuVx, 2016-06-12:
OK, I will investigate on Monday because right now I'm not sure.

digitalentity, 2016-06-13:
Current code is able to map stick command to 90 deg tilt without any problem, however the problem with 90 degree ilt is not the angle inclination itself.

At 90 deg tilt euler angles suffer the effect known as gimbal lock when one axis of freedom is lost. I.e. at 90 deg pitch there is no way to distinguish between roll and yaw anymore. We have quaternion IMU that is free from this effect but PID controller still use Euler angles.

We can make PID controller use quaternion algebra as well but this will make PID controller very taxing on CPU without giving us much benefit - navigation doesn't operate at extreme angles. This, however, doesn't mean we shouldn't try :smile:

DaleKramer, 2016-06-13:
@digitalentity  I believe that I understand the issues when using accelerometer output, as a couple years ago I wrote an iPhone app that provided the feedback for a pilot of a full scale airplane to follow a parabolic zero G flightpath within a defined 'G' error band on all 3 axis.  I found that the sensitivity of each accelerometer axis followed a sine wave. When approaching the 'flat' part of the sine wave for one axis, I just started to use a different axis to increase the sensitivity.  I have not researched qarternion.angles, but I assume that they must use some sort of axis combination.  Just to deal with this gimbal lock problem at 90 pitch angle, would not the easiest thing to do be to start combining the feedback from a different axis as 90 degrees in pitch is approched.  In this case it is not really need to handle ALL transitions that make a gimbal lock, just the lock at 90 degree pitch case.  But, then again, if the code handles all cases you would never have to think about this again.  Am I misunderstanding what causes gimbal lock?
I appreciate your willingness to discuss and perhaps help me with this issue.  Some people think I am a little too far out of the box but that is just me and I really was hoping not to have to 'learn' how to precisely achieve what I desire and end up doing it myself (I believe that current Cleanflight code handles 90 degree pitch in angle mode so that would be my start but on viewing the relevant code section, I see too many differences for me to easily deal with)! :)
I see that my $5,000 comment here may have been to my detriment but I will respond to that elsewhere.
Your INAV work and open sourcing is invaluable and very appealing to me.  It provides a great base for people like me to learn the nuts and bolts side of this amazing revolution in aviation without having to purchase components costing tens of thousands of dollars and having no ability to make it do what you really want without another huge expenditure.
Thanks,
Dale

digitalentity, 2016-06-13:
You should probably build a cheap and resilient 250-sized machine to crash-test your ideas. Current code won't work at 90 deg pitch - the closer you get to gimbal lock condition the more roll axis tend to start jumping around +180 and -180 degrees.

To have fully operational pitch of 90 deg we need quaternion PID controller.

martinbudden, 2016-06-13:
@DaleKramer , this youtube video [Sensor Fusion on Android Devices: A Revolution in Motion Processing](https://www.youtube.com/watch?v=C7JQ7Rpwn2k) is a good overview of how sensor devices can be combine and also has a section on gimbal lock and ways to get around it. Overall I think it's a very good video.

[As an aside, there was a slide containing something like "integration of noise = drift" which gave me an Aha! moment in a totally different area - I suddenly understood why genetic drift occurs - reproduction is a noisy process, evolution is a form of integration: result = genetic drift!]

DaleKramer, 2016-06-13:
@martinbudden , Thanks, as soon as I saw that sword I realized that I had seen that video before but for a different project :)
Wow, look at that, at 45 minutes, quaternion transformations can be implemented in one line of code, should be easy ;)
On the aside, neat theory but lets hope genetic drift is not a result of double integration.

DaleKramer, 2016-06-13:
@digitalentity , From what I understand, the betaflight guys are not even using the accelerometer in their PID controller.  If they use angle mode (heaven forbid), they have to turn their accelerometer on.  This says to me that Angle mode should not necessitate quaternions in the PID controller, what am I missing?  Or maybe their code still does not handle 90 degree angle mode.
I am building a tethered test stand for my vLazair multicraft and plan to tune and test all but transitions to and from horizontal flight in a 'safe' environment.  Unfortunately, I do not think building a 250 sized machine is possible to test the transitioning behavior.  That will be done in the 1/4 scale model after it is stable and hovers well on the tether rig.  Then it will likely progress untethered flight by taking off vertically, go high and experiment with transitions.  Angle mode at 90 degrees is a feature that I am pretty sure I need before I even attempt transitions.
I hope I am not scaring you off :)

DaleKramer, 2016-06-13:
I just thought, how does INAV PID controller handle gimbal lock, Can't you do loops, rolls and recover from unusual attitudes? Has to remain stable and pass through 90 degree situations doing those. Why does PID need accelerometers in acro mode (pilot corrects the drift automatically).  Don't modes that use the accelerometer to maintain a precise attitude just send some sort of bias signal to the PID loop so that the loop corrects to a certain angle and then maintains zero rates in the gyros with the stick centered.

digitalentity, 2016-06-13:
@DaleKramer 
You are misunderstanding how the FC control loop works. 

At lowest level there is a gyroscope that detects rotation on 3 axis. Rate PIDs stabilise **rotation rates** about these three axis. This is the only control loop that's present in ACRO mode.

At higher level there are **angles of rotation** about these three axis - these are effectively Euler angles defining orientation of the aircraft in space. Current ANGLE controller works by comparing these angles as estimated by IMU with angles requested by pilot and sets **rotation rates** for lower (rate) control loop.

The problem here is the boundary condition - gimbal lock, where actual position in space (90 deg pitch) may be defined with infinite number of Roll/Yaw angle pairs. We can improve this by doing the following:

1. Translate pilot requested angles into desired orientation quaternion.
2. Find a rotation quaternion represating a rotation between desired and estimated orientation
3. Decompose the calculated rotation quaternion to 3 components of rotation to feed to rate controller.

I'm not a mathematician so my understanding might not be entirely correct, but you can get the idea.

DaleKramer, 2016-06-15:
I am am the first to admit that I am not well versed in PID code intricacies, Thanks for trying to clarify!

On the bright side, when I was code scanning, I see that there is quaternion usage in imu.c, maybe this is closer to reality than I had hoped :)

Again, thanks everyone, this is a great community!

DaleKramer, 2016-06-15:
In the meantime, from what I can tell, the current source for INAV 1.1.0 seems to already allow requesting angles greater than 50 degrees, the problem is that when you actually use angles near and including 90, the craft will become unstable due to gimbal lock.

In [https://github.com/cleanflight/cleanflight/pull/1180](https://github.com/cleanflight/cleanflight/pull/1180) , @ctzsnooze suggests that in his experience things as pretty stable up to 80 degrees.  

That 80 experience limit may be enough for me, only testing will tell.

However, there is still an issue in that the current INAV 1.1.0 source will not scale the max stick travel to max_angle_inclination.

So, could someone confirm that these code changes, in pid.c, would scale the max stick travel to max_angle_inclination (I lifted this basic code from Boris's suggestion in the above link):

```javascript
//removed by dale    const float angleTarget = pidRcCommandToAngle(rcCommand[axis]);
//removed by dale    const float angleError = (constrain(angleTarget, -pidProfile->max_angle_inclination[axis], +pidProfile->max_angle_inclination[axis]) - attitude.raw[axis]) / 10.0f;
//dale added next line
const float angleError = (constrain((float)rcCommand[axis] * ((float)pidProfile->max_angle_inclination[axis] / 500.0f), -pidProfile->max_angle_inclination[axis], +pidProfile->max_angle_inclination[axis]) - attitude.raw[axis]) / 10.0f;
```

DzikuVx, 2017-01-31:
Some work done in #1077

digitalentity, 2018-03-09:
First step towards fixing this https://github.com/iNavFlight/inav/pull/2894

closed 2018-05-27 by stale[bot]