iNavFlight/inav #274: Implement F4 target support

martinbudden, 2016-06-13:
This is a list of tasks that need to be completed to support F4 targets on iNav
- [x] Move `pmw_init` and `timer` changes to `target` directory files
- [ ] Create F4 branch
- [ ] Merge in @ledvinap 's `exti_io` code, see PR https://github.com/betaflight/betaflight/pull/428 and [betaflight link](https://github.com/betaflight/betaflight/pull/428/commits/3787a435341b59f1c17320efdee1dbe63a5e110e) and PR https://github.com/cleanflight/cleanflight/pull/1421
- [ ] Update Makefile - in progress
- [ ] Add STM32F4 target linker scripts
- [ ] Update sensors and IO
- [ ] Update device drivers to STM32F4 as necessary
- [ ] Update main, config and startup
- [ ] Add STM32F4 targets, at least REVO
- [ ] Add STM32F4 VCP support
- [ ] Add USARTS 4,5,6 
- [ ] Add STM32F4 libraries

See also https://github.com/betaflight/betaflight/pull/452/commits for and PR https://github.com/iNavFlight/inav/pull/264 for useful commits.

blckmn, 2016-07-03:
@martinbudden Let me know if you need any help? I can always work on BF for anything missing and you can pull it in here.

martinbudden, 2016-07-03:
@blckmn , thanks for the offer. I don't think there are any blocking issues at the moment, but if I find any I'll ask for help. At the moment its just a question of working through the tasks. There are still some betaflight bugs outstanding, especially the F4 ESC passthrough - so your continued work in that area is needed and appreciated.

digitalentity, 2016-08-29:
I believe all is done here. @martinbudden can you confirm?

martinbudden, 2016-08-29:
I need to double check.

martinbudden, 2016-08-30:
@digitalentity , I've now checked through all these and there are, as far as I can tell, no F4 related changes that need to move over from betaflight.

However betaflight has fixed some bugs in the BLHELI passthrough code and also has some enhancements to the LED STRIP that are not yet in iNav. Do you want me to move these over for the next release candidate, or do you want to wait for 1.3?

digitalentity, 2016-08-30:
@martinbudden we have no bugs for them but they can be considered significant bugfixes. If possible please pull the code to INAV for next release candidate, thanks!

martinbudden, 2016-08-30:
Closing this. I will create separate PRs for the LEDSTRIP and BLHELI passthrough fixes.

closed 2016-08-30 by martinbudden