iNavFlight/inav #279: wiki suggestion: Guide for mission planing

skaman82, 2016-06-14:
I just startet to look into waypoints stuff a week ago and very quickly run in to some problems.
I can't find a guide how to setup things and things that I need for it. I am a mac OS user (no pc, no android devices) so information is nearly non existent.

So please, create a wiki (windows/android/osx/idevices in mind) how to get things going.
• What hardware is needed to monitor the quad during flight? (BT/3DR and so on)
• How to configure a mission and where is the GUI for that? (Win/Mac/...)

Can't find a GUI for MacOS, so I gave up on playing with missions for the moment.

stronnag, 2016-06-14:
I keep meaning to do this, from the numerous private emails I end up writing in the subject.

skaman82, 2016-06-14:
@stronnag can't wait for it!

skaman82, 2016-06-14:
It would also be great if someone could include GCS and a mission setup tab into configurator. 
I can help to make the interface for this.

stronnag, 2016-06-14:
wiki page started https://github.com/iNavFlight/inav/wiki/iNavFlight-Missions. This is work in progress.

skaman82, 2016-06-14:
Great, thank you. From the looks of it there is currently no solution for me to get mission planing working :(. I hope we can include this into configurator so we don't have the issues of different platforms.

stronnag, 2016-06-14:
I think MAVLink (for 1.12) will make other tools available, some of which are available on macOS (e.g. QGroundControl).

skaman82, 2016-06-14:
Thanks QGroundControl looks actually quite nice - looking forward to mavlink then :)

DzikuVx, 2016-06-14:
Question about MAVLink... It will be full MAVLink? This topic has somehow skipped me

stronnag, 2016-06-14:
It is not full MAVlink. AFAIK from when I tested the CF implementation with mwp, it is enough to do mission monitoring, and could easily be extended to support mission upload (I don't believe this is yet implemented on the FC side).

DzikuVx, 2016-06-14:
I've allowed myself to add a few links to my web page (QuadMeUp) with howtos and tutorials.

skaman82, 2016-06-15:
@stronnag this sounds promising for OSX/iOS users, at least after MAVLink is supported: https://github.com/mavlink/qgroundcontrol/issues/3573#issuecomment-226251791

digitalentity, 2016-06-15:
At the moment MAVLink support is going to be one-way (telemetry only). In future we might also implement waypoint protocol via MAVLink telem.

skaman82, 2016-06-15:
*sadface* :)

stronnag, 2016-06-15:
That is what the wiki page currently states.

skaman82, 2016-06-15:
@stronnag as far I can see there is no mention of the qgroundcontrol project

stronnag, 2016-06-15:
https://github.com/iNavFlight/inav/wiki/iNavFlight-Missions#telemetry-protocols

skaman82, 2016-06-15:
@stronnag sorry, my bad

oleost, 2016-11-22:
Guess this issue can be closed?

closed 2016-11-22 by digitalentity