iNavFlight/inav #280: Poshold not holds position after movement

skaman82, 2016-06-14:
Tested the latest release some days ago and have noticed something in POSHOLD mode. For filming its best to fly in poshold mode and the quad should stick to its position if the sticks are released - just like a phantom. The problem I noticed, is that if I enable poshold and move the quad 1-3 meters and release the sticks, it always wants to return to its old position. If I move the quad for longer time and more distance, it stays there. 

Does anyone had the same experience?

digitalentity, 2016-06-14:
@skaman82 
This is odd. Do you have blackbox recording of the issue?

skaman82, 2016-06-14:
Sadly no, the fc is in a closed plastic Walkere QX350 bodyshell, so it is difficult to hook up a openlog to it.

digitalentity, 2016-06-14:
Not good. Maybe somebody can reproduce the issue and post logs...

kronosol, 2016-06-20:
Mmh,.. it seems like the Inav code always take the last position where you centered your sticks.
So if you´re flying fast and center back your sticks to mid, the Quad have much speed left.
The code will break the Quad and fly back to the position you centered your sticks.
So in case you do some aggressive and short pitch on POSHold, your Quad will go for 1-3m, break and go back to that position you centered back to mid, which is nearly on that positon you start.
Maybe this helps? ;)
Btw: The POSHold is really aggressive, sometimes it would be great to have more like a smooth breaking. It´s nearly impossible to do a quiet and smooth break in POSHold at the moment.

theArchLadder, 2016-06-26:
@kronosol Not true, inav tries to calculate stopping distance when sticks are centered and sets the poshold there. Tune with `nav_pos_i`

@skaman82 This sounds more like a tuning issue to me, try a higher `nav_pos_i`, up to 200 if needed, that determines how long the stopping distance should be. Also try `nav_user_control_mode = CRUISE`, that way your stick controls speed instead of bank angle, and it feels much better, smoother and more accurate control imho.

kronosol, 2016-07-14:
@theArchLadder Hey sorry lost an eye on this issue here.
But thanks for your tipp with the cruise mode on PosHold. 
I´ve tested it today and I was like "whaaaat", so much better behavior of the Quad.
I´m really wondering why cruise mode isn´t default setting...? :(

theArchLadder, 2016-07-14:
@kronosol See #351

skaman82, 2016-07-17:
just tested the CRUISE mode. It is in fact a lot smoother and I like it much more then ATTI. But the cruise mode is a bit slow. I was able to move the quad only 10km/h max. Ist there a variable to speed it up just a little?  Is it nav_manual_speed?

theArchLadder, 2016-07-17:
@skaman82 Yes but it also seems to be limited by `nav_max_speed`. Try and raise both of them to like 900 (3x default) and it should be faster.

@digitalentity Is it a bug or a feature?

kronosol, 2016-07-17:
I´ve set *nav_manual_speed = 2000* and *nav_max_speed = 500*, flys save but only around 2-4m/s max. Also there is a issue with yaw. @digitalentity @theArchLadder 
Maybe increasing *nav_manual_speed* also effects on yaw rate?
If I enable Cruise Mode I can turn on yaw more like 360°/s on full stick deflaction, which seems crazy high. 
If I do the same with *nav_manual_speed = 500* speed on yaw is normal like on ATTI mode..
**Interesting:** If I enable RTH with *nav_max_speed = 500* it´s even faster than Cruise Mode with *nav_manual_speed = 2000*.

digitalentity, 2016-08-29:
@skaman82 is the issue still relevant?

closed 2016-11-21 by digitalentity