iNavFlight/inav #282: Readme needs updating

martinbudden, 2016-06-14:
The readme is all about Cleanflight - it should be about iNav!

digitalentity, 2016-06-14:
You are right! I'm always busy writing code - usually don't have much time to write docs :smile:

DzikuVx, 2016-06-14:
OK, done in a588ee08b96dcbba54e903b9c9b596a7e068ded8
I also allowed myself to change few sentences, remove baseflight migration

martinbudden, 2016-06-15:
I don't think we should close this yet.

The readme doesn't really give a summary about what iNav is all about. The first line is:
"Clean-code version of baseflight flight-controller - flight controllers are used to fly multi-rotor craft and fixed wing craft." It goes on to talk about software development practices and then has a big list of features none of which are to do with navigation.

I'll mark this issue as "in progress" and people can make improvements as and when they have time.

DzikuVx, 2016-06-15:
Yeap, you are right... help welcome

PR #284 (2016-06-14): docs update
Cleanflight replaced with INAV in docs
Fixes #282