iNavFlight/inav #287: Design a FC for INAV

digitalentity, 2016-06-15:
I think it's time to think about designing a FC specifically for INAV. 

Features I want to have on this FC:

1. Future-proof design with an F4 (or even F7) CPU
2. 8 outputs supporting 8/0, 6/2, 2/6 motor/servo configurations
3. Onboard baro (MS5607 or BMP280)
4. Optional onboard NRF24 receiver (placeholder for a tiny daughter PCB)
5. External I2C connector (compass)
6. External SPI connector (opflow)
7. SD-card logging
8. Hybrid 4-pin connector to PWM/SBUS receiver (3.3V + 5V both available), same pin used for PPM/SerialRX
9. USB VCP
10. At least 3 extra UARTs for GPS, MSP, Telemetry
11. 5V-level LEDSTRIP connector
12. Analog inputs for RSSI, VBAT, CURR and an extra for optical rangefinder

What do you think?

skaman82, 2016-06-15:
13. Sonar Pads

Linjieqiang, 2016-06-15:
Wonderful!I have always been concerned about this project.I am designing a FC board now and ready to flash inav F4 firmware,use STM32F4RGT6.I suggest why not use Can Bus Hub like Dji to connect sonar,opflow,gps,osd or even more sensors and devices?It will save hardware resources.

DzikuVx, 2016-06-15:
What about factor? Probably keeping 30-something (30.5mm hole spacing) would be impossible. I'm fine about that, Pixhawk size would be totally fine in my case. I would even prefer that.

It would be also very nice to have SoftSerial not conflicting with RSSI and ADC

digitalentity, 2016-06-15:
@skaman82 
Done!

@Linjieqiang 
I'm thinking about being able to connect a sensor hub via serial port - this way sensor hub will be usable even with older FCs without CAN. I don't want this design to be bound to one hardware only.

@DzikuVx 
I'm going to keep 30.5mm hole spacing but the overall form factor is going to be larger (longer board). USB on the side, connectors on front and rear edges.
I'm not going to make SoftSerial available on this board - F4 CPUs have enough hardware UARTs. 

I'll design the board in such a way that it would be possible to use all (or almost all) features at once without conflict.

skaman82, 2016-06-15:
1+ for 30.5mm hole spacing

DzikuVx, 2016-06-15:
I would be very happy with 5 Hardware Uarts

digitalentity, 2016-06-15:
@DzikuVx this might be impossible. I'm going to use UART2 RX for PPM/SERIALRX input so it's going to be UARTS 1,3,4,5

DzikuVx, 2016-06-15:
I know I am an special case here. I have OSD via MSP, another msp for telemetry, GPS, FrSky S.Port and Blackbox. All 5 possible Uarts taken on my Sprf3.
Probably will switch osd to ltm and share it with telemetry via some Small arduino to free one uart

oleost, 2016-06-15:
Just a question. 

Hardware UARTS are fine, yes. But Softserial has proved very usefull in many cases.

With an F4 cpu compared to and F3 CPU could we overcome some of the limitations?
Or with the F7 cpu?

Not a hardware developer, nor a software. Just thinking with a more powerfull CPU maybe softserial could run at 100k baudrates?

bk79, 2016-06-15:
I would use mpu9250 to have compass with no extra wire and an integrated
bec so to avoid having a different board and all the io pin out avaiable

skaman82, 2016-06-15:
@bk79 External compass mounted far away from the wiring is always better then a internal one.

DzikuVx, 2016-06-15:
And this is why I like the idea of no magnetometer on this board

bk79, 2016-06-15:
As long as it is far from ac power source (escs)  it's ok and unless you
have  a 250 size which is intended normally for racing i can assure it
makes really little difference to have it inside or outside the fc shell if
you like i can attach the reading coming from a sparky connected to an hexa
320  and the reading coming from a cc3d connected to an 400size with ext
compass of course in hoovering conditon for both that means low currents
radiating

bk79, 2016-06-15:
Just as addition the bigger is the angle between escs wire and compass the
higher will be the induction current from the esc wires since their
radiation field is radial to the wire and normally the wire are on the same
plane of the fc which means low mutual induction

Linjieqiang, 2016-06-15:
@digitalentity oh,I see.Your idea is great.What processor chip do you want ?What about STM32F4RGT6?I think it's ok.Anyway,look forward to your work.

digitalentity, 2016-06-15:
Vibration damping of the sensor board is a good idea but it will make the FC harder to manufacture. You can easily soft-mount the FC itself.

bk79, 2016-06-15:
I think thata mix  spracingf3 deluxe and evo with sdcard reader is a good
starting point but with f4 processor

dan557, 2016-06-15:
If most people are going to use a osd on one of the uarts why not build a
max chip into the FC and with a f4 or f7 processor there will be plenty of
resources to support it .
osd support is currently being added to cleanflight

digitalentity, 2016-06-15:
With HD systems emerging having an analog OSD is not the best idea. Also
what if one will want different OSD? If possible I'd like to avoid mixing
flight control and vision.

Linjieqiang, 2016-06-15:
ok.When do you start to design?open source schematic?I also think fc is only flight control,not mix other,such osd……I am waiting for learning your optional flow code.

DzikuVx, 2016-06-15:
I also do not like the idea of mixing FC with OSD.

digitalentity, 2016-06-15:
What do you think, maybe make first version F3 based?

skaman82, 2016-06-15:
@digitalentity what would be the advantage, compared with a SPracing F3 (my board of choice at the moment) - except SD Card support?

digitalentity, 2016-06-15:
A very brief connector layout
[image]

This may even fit into square board with 30.5 mounting holes. But more likely it will be a rectangular board with regular thru-hole connectors on both sides (motors on one, RX/extras on the other)

digitalentity, 2016-06-15:
@skaman82 for example gyro on SPI bus, direct support for SPI external hardware (optical flow), more hardware UARTs + USB for Configurator

digitalentity, 2016-06-15:
Also, what about exposing USB as through-hole connector to be able to easily move it to the outside of the aircraft?

digitalentity, 2016-06-15:
Laying out the board is going to be one hell of a task :smiley:

bk79, 2016-06-15:
Spi needs a dedicated pin for each device exposing one pin only means
having a dedicate connection only for the optical flow and using f3 would
loose any benefit compared to f4 since spracing pro deluxe has everything
with less than 40$

digitalentity, 2016-06-15:
We can move CAN to the side (or lose it at all) and have 3 \CS pins for SPI meaning 3 possible external devices to connect (SPI RX + OpFlow + ...)

digitalentity, 2016-06-15:
And I agree - using an F3 chip is not a very good idea

bk79, 2016-06-15:
F4..ok f3...still not much gain compared to other boards

digitalentity, 2016-06-15:
@bk79 what would you consider "much gain"?

skaman82, 2016-06-15:
• More hardware UARTs is always nice! 
• I have nothing against a F3 version if it makes the CF cheaper. I personally don't like spending more than 30$ on a FC.
• About the usb: I have to solve that problem constantly (because SPracing boards have the USB on the rear side. I just take a micro usb connector and connect it to a cheap USB breakout board. This way I can place the USB everywhere I like. It is pretty easy. If you placing the USB on the side, there is no need to have extra pins for that - they take away some space to.

digitalentity, 2016-06-15:
Actually STM32F303RC costs roughly the same as STM32F405RG, so it's going to be an F4 CPU after all. The problem is UART hardware inversion which F4 chips don't have.

digitalentity, 2016-06-15:
Actually $30 price for this FC is marginal, BOM+PCB+assembly will cost ~20-30$ if produced in low quantities. What price do you think would be reasonable for this FC?

skaman82, 2016-06-15:
I just talked about me (personally). I see people spending 40-50$ for this easily.

digitalentity, 2016-06-15:
Personally I don't like spending more than $30 on an FC as well, but I don't mind prices up to $50-60 for a decent product :smile: 

Actually, a dead simple board w/o baro, w/o CAN and w/o SD-card might cost even under $20. I need to do a proper design to be able to estimate BOM and assembly prices.

skaman82, 2016-06-15:
The original SPracing F3 was to pricey as it came out (it was about 65€). That was a deal-breaker for most and that is mainly the reason why spracing boards are not so popular. It became more popular since china started to clone this FC and sell it for about 20-25 €.

DzikuVx, 2016-06-15:
Suggestion that came to me after todays helping fellow... If 3.3v is supposed to be available on a connector, use 2 ldos. One for cpu, one for external pin. And maybe polymer fuse as well

bk79, 2016-06-15:
@digitalentity benefit from a faster cpu are huge quaternion can be used
sdio for sdcard is there 4 usart various i2c and three spi...

Suggestion that came to me after todays helping fellow... If 3.3v is
supposed to be available on a connector, use 2 ldos. One for cpu, one for
external pin. And maybe polymer fuse as well

—
<https://github.com/iNavFlight/inav/issues/287#issuecomment-226314901>, or mute
the thread

rustyoz, 2016-06-16:
I would love a board with a separate 5v rail on the motor/servo outputs with pins for an external input. I have a tilt quad and using a seperate 5v bec for servos or even high voltage servos would be a plus.

martinbudden, 2016-06-16:
My thoughts on the iNav FC.

I'm not a hardware guy, so my thoughts should be taken with that in mind.

### Thoughts on the spec

1) Future-proof design with an F4 (or even F7) CPU
Leading edge, not bleeding edge, so F4 not F7. With F4 iNav can share code with Cleanflight and betaflight. With F7 it's on it's own. And F4 rather than F3 since the FC is for iNav and iNav will add more navigation calculations in future.

2) 8 outputs supporting 8/0, 6/2, 2/6 motor/servo configurations
Yes, definitely 8 outputs. 6 is not enough for many iNav uses.

3) Onboard baro (MS5607 or BMP280)
Yes.

4) Optional onboard NRF24 receiver (placeholder for a tiny daughter PCB)
Most definitely. Note that the NRF24 requires a stabilised 3.3V power supply, and behaves erroneously if the voltage drops too low, so ensure your onboard 3.3V regulator is sufficiently powered.

5) External I2C connector (compass)
Yes. And external I2C connector can be used for other devices as well. Eg sonar, optical rangefinder etc (see further comments later).

6) External SPI connector (opflow)
Yes. And external SPI connector can be used for other devices as well. Consider having more than one NSS connector available. Other SPI signals can be shared.

7) SD-card logging
Yes.

8) Hybrid 4-pin connector to PWM/SBUS receiver
Yes.

9) USB VCP
Yes.

10) At least 3 extra UARTs for GPS, MSP, Telemetry
Yes, but see later comments on UARTs.

11) 5V-level LEDSTRIP connector
Yes.

12) Analog inputs for RSSI, VBAT, CURR and an extra for optical rangefinder
Not entirely. See later comments . In brief VBAT not required if there is an onboard BEC, and optical rangefinders seem to use I2C.

13) Sonar connector
Yes.

It seems optical rangefinders use I2C. Both the [teraranger](http://www.teraranger.com/products/teraranger-one/)
and the [lidar-lite](http://www.robotshop.com/uk/lidar-lite-2-laser-rangefinder-pulsedlight.html) can connect via I2C. (Now that Garmin has bought PulsedLight, it's unclear whether the Lidar-Lite will continue to be available though)

OSD has been mentioned in the thread. My view is that the "natural" place to integrate the OSD is in the VTX, not on the flight controller.

### Sensors

Gyro should connect over SPI, not I2C

Consider using Invensense's temperature stabilised gyro the [mod-1600](https://www.invensense.com/products/mod-1600/) 

I believe this is used on the [pixracer](https://pixhawk.org/modules/pixracer).

If you don't use the mod-1600 my preference is to use the [MPU-9250](https://www.invensense.com/products) with built-in compass but have the option to add an expernal compass.

### Thoughts on serial ports

While I think currently you need to maximise the number of serial ports, I think in future this will be less important.

We are already seeing devices that used to be connected via serial ports using different options:

Team Black Sheep's BST(Black Sheep Telemetry) uses I2C rather than serial. I think we will increasingly see I2C used for telemetry.

We are seeing begining to see OSDs connected via SPI, and VTXs with integrated OSD.

[... 45 lines cut ...]
This means having at least 4 FET-buffered PWM outputs on the board.

This is not unprecedented. The SPARKY 2 flight controller has 4 buffered PWM outputs that can be used to directly drive brushed motors.

The KISS flight controller has buffered PWM outputs, although in this case I think they are for performance reasons (and KISS know a little bit about performace).

The [AlienFlight F3 brushed-octo](http://alienflight.com/f3-brushed-octo/) FC uses 4 dual-channel MOSFETs for its PWM outputs (it uses Dual N-Channel Fairchild FDMB3900AN MOSFET) so 4 PWM outputs could be buffered on the iNav FC using just two of these dual-channel FETs - that should not take up much board space.

Personally I'd be in favour of buffering all eight PWM outputs. I think buffering the PWM outputs may bring other advantages - performance and the ability to drive high power servos.

Here's what I'm thinking: someone could buy a cheap brushed-motor quadcopter, say a Syma X5 or X8, or a Kai Deng K70. They could then replace the original flight controller with an iNav FC, add an NRF24 daughterboard and a GPS unit and they have a very cheap quadcopter that supports navigation. They could probably even use the transmitter that came with the quadcopter presuming it's protocol was supported, so they wouldn't even need to buy a transmitter.

Brushed motors are pretty well universal on sub-120mm copters, and I quite like the idea of a navigable small copter like that as well. [And for those of you who haven't build a brushed quad - their perfomance can be surprisingly good and the flight fast and agile.]

#### Onboard BEC

I'd like to have an onboard BEC. This is just so much more convenient. BEC only needs to support 2S-4S. For a 5S or 6S aircraft the FC can be powered from the balance lead.

[As an aside, I generally wire my quads that way - power the FC from the balance lead, power the ESCs from the XT60 connector. That way you can power up the FC without powering up the ESCs. That means you can do some setup operations safely without removing the props. It also reduces (slightly) the electrical noise from the ESCs to the FC. Not my idea - I first saw it on the setup pages for the Brain FC.]

The onboard BEC also makes it easier to support brushed quads. (Note, no 3.7V to 5V boost is required. The way to support 3.7V brushed motors is to use a 2S battery (or two 1S batteries), and power the FC from both cells, and then use the one cell to power (say) the front two motors and the other cell to power the back two motors.)

#### Digital outputs

I'd also like to see a couple of digital output pins (ie ON/OFF, not PWM). These could be used, for example, switch on LEDs if an LEDstrip was not being used, or to control a camera. These are on my "nice to have" rather than "essential" list.

DzikuVx, 2016-06-16:
My note on Martin's suggestions:

1. Build in BEC: those things when with proper max. load tends to be big and heavy. And costly. And if it burns, one would have to usually throw whole board away. Instead I like the idea of separate power line with a jumper that might allow to use high voltage servos
1. PWM outputs with MOSFETs to brushless motors: hmm... again, not a biggest fan in this case. While I'm totally fine with another FC with brushed support and without fancy stuff like SPI,CAN or even most of UARTs, putting all possible features on one board will make this big and costly. The biggest advantage of INAV is GPS. And somehow it's hard to imagine brushed motors carrying those
1. Digital outputs: nice, too bad software is not ready for them yet. And to be useful, they would require MOSFETs
1. PWM Connectors: I would stick with traditional ones because of servos

martinbudden, 2016-06-17:
@DzikuVx 

1. I've seen from @digitalentity 's comments elsewhere that he is not in favour of a built-in BEC, so I'm probably onto a loser there :-) However, for the record, I don't agree with your big and heavy comment - this may have been true in the past, but there have been developments in BECs. A number of "small" FCs include BECs these include the KISSFC and the LUX (which are both sub-36mm size - in that they are 36mm boards with the sides cut out. And the tiny FPV-PIKOBLX (which has 20mm mounting hole spacing) has a built-in 2A 5V BEC.
2. I find it very easy to imagine brushed-motor quads carrying GPS. The Syma X8 is a very popular 330mm brushed motor quadcopter and I bet loads of owners would love to upgrade to GPS. The iNav FC could allow them to do so, and with the NRF24 daughterboard they would not even need to buy a new transmitter. Depending on the iNav FC cost they could have a GPS enabled quadcopter with transmitter and camera for about £100. (Admittedly the transmitter and camera are both a bit rubbish, but those could be upgraded if required.) Rather than a priori dismiss the addition of FETs, I think we should let @digitalentity do a BOM and rough board layout - if they are too costly or there is not room, then fair enough - but let's not pre-judge the result.
3. Digital outputs "too bad software is not ready for them yet" - well that is something I can change :-)
4. PWM connectors. I think I have a solution that works for both of us, see below.

#### Possible option for PWM connectors

One of the interesting aspects of the SPRacing Mini flight controller is the addition of "fault lines" on the PCB, so that parts of the PCB that are not required can be snapped off. This could be used on the iNav FC for the PWM connectors, e.g.:

```
fault line ____GV12345678___ fault line
                 VVVVVVVV
                 GGGGGGGG
```

That way users who did not require the extra two rows could just snap them off.

The extra two rows could be outside the 36mm square size, so that snapping them off produced a 36mm square board.

sppnk, 2016-06-17:
And..what about an on-board OSD chip? The OSD tasks probably could be done by the main processor and the design still would be compact.

Then we would need also the conectors for camera and vtx (3.wire servo conectors, or may be 4-wire connectors, including audio).

DzikuVx, 2016-06-17:
@martinbudden 

* Nice FC you have found there. Nice step-down incorporated design. Too bad BEC is taking almost a quarter of one side (not counting pins). I have 2A "small" BEC in one of my drones, but even this small one is kind of "biggish" in terms of 36mm and somehow I would be afraid to really 2A from it
And since there will be a lot of pins in this FC, 36x36 might be just too small.
* I like faultline solution. With fault line board can be bigger than 36x36 in total. That might work.
* Digital outputs... Hmm... If they were MOSFETed and accept 500mA load that would be awesome.... With proper software support of  course

digitalentity, 2016-06-17:
**FET-buffered PWM outputs:**
If I have space on the board I might add FETs to buffer PWM outputs to allow brushless motors. If it will be possible I'll buffer all 8 of them. The problem here is BLHeli passthrough - buffering is going to kill it.

**Sensors:**
No I2C. Gyros will be SPI-bus. MOD-1600 looks promising. I'll try to get a sample and design a board to use it.

**PWM connectors:**
There are lots of tiny FCs and this is not going to be one of them. As we target also airplane flyers there will be no compromises - each servo/motor is going to be a 3-pin header connector. I might try a "fault line" solution though.

**OSD:**
I don't like all-in-one solutions. THe most logical place for OSD is VTX (or camera).

**Onboard BEC:**
As a replacable daughterboard - it's possible, but not as a built-in option. BECs burn out and I'll hate to replace the whole FC. Also, I like specialised solutions - dedicated BEC will handle higher loads, will be easier to replace and will reduce cost for those who don't need it.
I might add a 2-4S BEC to power FC from VBAT directly but this will be an option to power FC only, no servos, no external devices.

**Digital outputs:**
And your whish shall be granted :smile:

digitalentity, 2016-06-17:
It's not the code, it's the hardware. BLHeli requires a bidirectional link
between FC and ESC. Mosfet buffer is uni-directional in nature. If I won't
be able to make a hardware solution it's going to be either passthrough or
buffers.

martinbudden, 2016-06-17:
I see. If you can't make a hardware solution for BLHeli, then I suggest the best option is to buffer 4 of the 8 outputs. That way brushless quads can use BLHeli passthrough, and brushed quads are also supported. The downside is no BLHeli passthrough for hexacopters or octocopters, and brushed hexacopters and octocopters need to add additional FETs. But I think that is a reasonable compromise.

iforce2d, 2016-06-19:
Absolute must:
- USB
- blackbox capable
- 3 hardware UARTS
- I2C
- buzzer
- vbat
- 8 outputs

Really would like
- SPI (one is enough, maybe use configurator to specify some other pin as second device SS pin, like we do for softserial now)
- 8 inputs

Don't care
- CPU type (although 3 hardware UARTs implies F3 or above)
- onboard baro
- 30.5mm spaced mounting holes
- size larger than 36mm square
- led strip stuff
- RSSI
- curr
- sonar

Prefer not to have
- onboard mag
- OSD stuff
- brushed motor stuff

Really don't want
- those little plastic plug connectors.... yuk

The main thing is to have lots of connection options, that way other parts can be added as needed and the cost can be kept reasonable. Most people will not be using iNav on a little mini-quad, so I would think the wriring space needed to add, for example, a separate OSD or whatever will usually not be an issue.

oleost, 2016-06-19:
Regarding the positive on servo/motor outputs it would be nice with just and simple jumper solution or two solder pads one needs to bridge to get it connected with the rest of the FC.

So one could power servo independently from the FC.

digitalentity, 2016-06-19:
That's exactly what I'm going to do

kronosol, 2016-06-20:
Hey some thoughts here,.. @digitalentity 
A nice thing to have would be a onboard bluetooth module or something like that, instead of an internal MAG. It makes no sense to put a MAG on the FC board anymore as you will get much better results with an external MAG. (NEO-M8N/beginner Questions about toiletbowling..). For the case of setup a MAG on the FC, PLEASE! find a solution to switch off that MAG, like the way they did it on Naze32 REV6 with a litte line you have to cut.
Also the same for the baro,.. and in case of setup a internal baro, please put it on the bottom side, so it is covered from sunlight and foam will not block the LED´s.
SD-Card holder just takes to much space,.. In my opinion a memory chip will be just enough? I just need a Blackbox log with a max length of 30min. Just let the code override the memory all the time, so at a failsafe/crash case you can see what happened. You really don´t need the logs of the last 3 months on your quad,..
To create a successful FC we should keep it as simple and tough as possible!
(In my mind..)
You need a GPS/MAG? Plug it in!
Your wiring is messy and USB is hidden? Use bluetooth!
You crashed? Look at the latest Blackbox Log files!
You´re sick on GPS flight assistance? Ok, Inav-FC is even light and simple enough for Racer Frames.
(...)
Btw: The final price should be max of 50€.
Btw2: Even I like the layout of the Openpilot CC3D Mini with these JST-Connectors, we should focus on established connections :)

digitalentity, 2016-06-20:
Acc/gypo chip: MPU9250 via SPI bus. Compass built-in but can be disabled in software

Back side of standard 36x36 board is very tight. I think it's going to be either SD-card or nRF24 daughterboard. Round-robin writing is not possible with supported blackbox architecture and SDCard give us more options - like storing missions.

iforce2d, 2016-06-20:
What exactly does a nRF24 daughter board look like? As a regular user of the nRF24, I think it's good enough if there is an easy to access SPI header. Depending on the form factor of the nRF module, I often want to mount it differently.

digitalentity, 2016-06-20:
You'll have SPI header on new FC.

martinbudden, 2016-06-20:
@digitalentity @iforce2d , I concur: SPI header is sufficient for nRF24, in fact it allows the user to choose their own nRF24 module. A 3.3V output pin is also required, and preferably this should be right next to the SPI header.

krzysztofmatula, 2016-06-21:
> 8 outputs supporting 8/0, 6/2, 2/6 motor/servo configurations

To control gimbals up to 4 servo/forward PWM outputs could be useful (pan, tilt, gimbal mode, video switcher). So, 4+4 setup, or even 6+4 (or 8+2) option would be nice (not necessarily available on pin headers, maybe just some soldering pads for the additional 2?)

Is it technically possible on the chips being considered?

I vote for the SD card slot.
On board compass - ok, but allow it to be disabled in software.
USB slot on the side - yes.

DzikuVx, 2016-06-21:
@krzysztofmatula motor/servo is a delicate issue. From the code point of view it's a huge mess. But I concur that 4/4 would be very useful.
You might want to take a look at #261 . With additional hardware PCA9685 (~ $4) and this change it will be possible to run up to 16 servos via external driver. So, airplane (5 servos) with gimbal and PPM decoder? Why not... On almost any flight controller

csarka, 2016-06-21:
These are all great thoughts. I think we're going to end up with something great!

My two-cents worth: 
In the $40-$50 range is worth it.
I agree with iforce2d (please don't use those little plastic plug connectors).  All the connects should be the 0.1in standard through hole for solder pins.
It should have the standard square 30.5mm spaced mounting holes, however it's ok if the board sticks out further.
The possibility to mount the USB connection externally is a nice idea.

For skipping boot sequence (when we mess it up, haha) what about a surface-mounted button, or at least wired to the standard 0.1in pins.  (No tiny solder pads.)

[86 more replies]

closed 2016-11-21 by digitalentity