iNavFlight/inav #293: RC Yaw is reversed in current master

stronnag, 2016-06-17:
Some change between 4db8d9c and 19887ef causes RC Yaw to be reversed (at least for PPM).

This is somewhat disconcerting and probably not intended.

digitalentity, 2016-06-17:
Thanks for reporting. Is it RC channel or direction of rotation of the aircraft when you command yaw?

stronnag, 2016-06-17:
It appears to be rotation of the aircraft. I move the stick expecting to yaw left, the machine yaws right. All other RC inputs appear normal (I can fly back 'home' on roll and pitch).

digitalentity, 2016-06-17:
Thanks. It's probably recent merge of rates in degrees per second. I'll
look into it when I get to my laptop.

digitalentity, 2016-06-17:
I believe this commit is responsible.
https://github.com/iNavFlight/inav/commit/9032cb846e75e5b42a8e32677226d66609811db1#diff-8424c55cddd22ef29138ff3ddee226a8R172
In code before that rcCommand was negated after lookup, that commit lost the -1 multiplier.
@DzikuVx is this intentional or a bug?

DzikuVx, 2016-06-17:
Yay, looks like we'lost -1 multiplayer for yaw somewhere there indeed.. Yaw has to be reversed. For next few days i'm unable to fix and test. Anyone else? @martinbudden maybe?

digitalentity, 2016-06-17:
I can fix it but I can't test it.

stronnag, 2016-06-17:
Fix it, I'll test it. I can only break stuff.

DzikuVx, 2016-06-17:
Maybe correct approach would be to negate at lookup table level? There is now a getter for yaw lookup that could do it if I remember correctly

DzikuVx, 2016-06-17:
My snartphone says it can be done in rcLookupYaw. Probably

stronnag, 2016-06-17:
My quadcopter says it would be nice it potentially aircraft breaking commits were not made to master until they have been flight tested.

digitalentity, 2016-06-17:
@stronnag When merging the #204 I was under impression that it's well tested in flight by @DzikuVx, didn't flight-test it myself, sorry :pensive:

martinbudden, 2016-06-17:
@stronnag , I agree - this commit should have been flight tested. Even a bench test would have picked up this problem.

stronnag, 2016-06-17:
No problem, there was no damage (other than to my pride from the "walk of shame" to recover the force landed machine). 

As a number of developers branch off master, it would be helpful if we could establish a policy of at least bench tested commits.

martinbudden, 2016-06-17:
@stronnag , years ago, when I was a relatively inexperienced programmer I had an experience which changed my attitude to testing profoundly.

I had a plumber install a new shower in my flat. He said he had finished the job, so I went in the bathroom and turned the shower on. It leaked. I couldn't believe he hadn't tested it. I realised that I and my colleagues at work sometimes had a similar lackadaisical to testing, so I changed my approach. [Needless to say, I never used that plumber again.]

DzikuVx, 2016-06-17:
Well, what can I say... But this is true, changes after resolving latest conflicts with Martin's changes were not flight tested, only bench tested. My bad.
pid controller changes were tested, I'm flying it

martinbudden, 2016-06-17:
@DzikuVx - no worries, we all make the occasional mistake.

DzikuVx, 2016-06-17:
@martinbudden still, one forgotten -1 a look what happens. I'm expecting more from myself :)

DzikuVx, 2016-06-18:
Sine it's fixed in #298 , I'm closing it

closed 2016-06-18 by DzikuVx