iNavFlight/inav #295: Workflow change

DzikuVx, 2016-06-17:
To prevent things like #293 I propose change in the workflow and introduce slightly simplified GitFlow methodology:

1. feature branches are merged to `develop` branch.
1. Non-feature-branched commits are done only to `develop`
1. `master` branch reflects latest release
1. `develop` is merged to `master` only when official stable release is published

I'm using this for a few years now and it greatly reduces number of strange things happening in `master`.

martinbudden, 2016-06-17:
Yes, I think this is a good approach.

I'd make one small change: call the branch `development` rather than `develop` (for the simple reason that betaflight has just adopted this approach and has called its branch `development` - for those of use working on more than one fork a simple change like that makes a big difference).

DzikuVx, 2016-06-17:
I can live with that :)

DzikuVx, 2016-06-17:
Also, for those who does not know it, here is why [you should not lie to Git](http://paul.stadig.name/2010/12/thou-shalt-not-lie-git-rebase-ammend.html) 

So, I have a suggestion to limit rebase usage. I know there are many dev who uses rebase very often because of "advantages" but well.. Right not I can not even find a commit when `-1` was lost because probably it was lost during rebase conflict.

martinbudden, 2016-06-18:
@DzikuVx , I am familiar with your link and disagree with some of it and think much of it does not apply to iNav.

I don't have time to write an essay about why I disagree and think it does not apply, but my disagreement is essentially based around two things:

1. Cleanflight is a massively forked project and we want to have the ability to freely move code between the forks, especially Cleanflight, betaflight and iNav
2. Cleanflight adopts the practice of running the test code against the rebased PR before it is merged, so the arguments around using git bisect don't apply, since every commit into the master passes the test code - it's not possible to do this if you don't rebase before the merge.

The actual underlying problem is that the test code is not working at the moment. The test code should have picked up this problem (and if it didn't one of the fixes to the problem would involve writing the appropriate bit of test code.)

Getting the test code working is dependent on issue https://github.com/iNavFlight/inav/pull/151 - _Migrate to Parameter Group config storage (catch-up with CF)_ which is the next thing I will work on after the F4 port.

martinbudden, 2016-06-18:
@DzikuVx , here's what Linus has to say:

http://lwn.net/Articles/328438/

> I want clean history, but that really means (a) clean and (b) history.
> 
> People can (and probably should) rebase their _private_ trees (their own 
> work). That's a _cleanup_. But never other peoples code. That's a "destroy 
> history"

It's worth reading the whole thing.

digitalentity, 2016-06-20:
When we're done with 1.2 we'll adopt this methodology

digitalentity, 2016-08-18:
With 1.2 imminent I'm declaring feature freeze on master. Only bugfixes will be merged until 1.2 is released.

After that we'll have a `develop` branch and master will be protected and updated only through pull requests from the `develop` branch (this doesn't include PRs that are already opened against `master`, however I encourage their respectiveauthors to re-open them against `develop` once it's created).

With some luck this will happen by e/o next week.

martinbudden, 2016-08-18:
Can we call the branch `development` rather than `develop`? I work on betaflight as well, where it is called development, and I can see myself forever typing the wrong name!

digitalentity, 2016-08-18:
@martinbudden `development` it will be.

digitalentity, 2016-08-19:
I've created `development` branch, please submit new PR's against this branch. 
Please DO NOT MERGE anything to development yet, until 1.2 is out development should be an exact mirror of `master` to avoid.

digitalentity, 2016-09-07:
1.2 is out. We can start changing workflow now.
`development` is now aligned to `master`, new PRs should be done against development branch now.
I think we need to re-open existing PRs against development branch as well.

stronnag, 2016-09-07:
Note the rel note still says rc5. 

Congratulations on final!

digitalentity, 2016-09-07:
> Note the rel note still says rc5.

Oops, thanks. Fixed.

DzikuVx, 2016-09-07:
You do not have to reopen PR. You can just change base branch

digitalentity, 2016-09-07:
Didn't know that, cool!
I've also created a label "DEV" to mark PRs made to development branch for the transitional period

DzikuVx, 2016-09-07:
I just changed all mine PR to use `development` as a base

martinbudden, 2016-09-07:
Brilliant!

And you just beat betaflight 3.0!

digitalentity, 2016-09-07:
I've just enforced that PRs must be up to date with base branch and passed through travis-ci before merging. This should make us break things less (hopefully) :smile:

DzikuVx, 2016-09-15:
I think we can close it since we started to use new workflow

closed 2016-09-15 by DzikuVx