iNavFlight/inav #300: Switching from GCS_NAV to POSHOLD does not use current position

iforce2d, 2016-06-19:
I have been playing around with the GCS_NAV mode to update the 'special waypoint' for a follow me functionality. I notice that when I switch off the GCS_NAV mode (returning to normal POSHOLD) the quad will keep moving to the 'special waypoint' location. I think it should instead set the current position as the desired position. One of the main reasons for having a distinct GCS_NAV mode was to avoid this behavior, because (to quote eziosoft) "The user may want to have PH ie in panic situation." #103

digitalentity, 2016-06-20:
GCS_NAV isn't a flight mode, it's an indication that PH position may be updated from the ground. Nevertheless, described behavior is a bug.

iforce2d, 2016-06-21:
Yes, I noticed that it's not part of the state machine. From an end user's point of view though, it looks and behaves just like a mode :)

andre92, 2016-07-01:
Maybe add poshold reset to gcs_nav when turning it off?

PR #392 (2016-08-09): Reset POSHOLD target when GCS modifier is reset
Closes #300
closed 2016-08-09 by digitalentity