iNavFlight/inav #301: Is AK8975 compass still supported?

krzysztofmatula, 2016-06-19:
I'm experimenting with an external CJMCU-86 module and the AK8975 compass. I'm connecting the module over I2C to my SPRACINGF3 deluxe.

Now the issue: whichever mag_hardware I use (0, 2, 3) it seems always the on-board compass is in use. 

I didn't verify if my AK8975 board is actually working, but I'd expect the "heading" to show some default value after setting mag_hardware=3 in case of failure. Instead it shows the readings from on-board HMC5883.

DzikuVx, 2016-06-20:
According to the code, AK8975 is used. Driver is present, code detection in place. Maybe there is something deeper. Can you confirm Cleanflight behavior with the same setup?

digitalentity, 2016-06-20:
It should work. I don't have an AK8975 board to test though

krzysztofmatula, 2016-06-27:
I've just confirmed identical behavior on cleanflight v1.13.
New issue: https://github.com/cleanflight/cleanflight/issues/2265

krzysztofmatula, 2016-07-08:
FYI: A response to cleanflight issue points the fallback mechanism, so in case of problems with MAG initialization another attempt is done with the DEFAULT mag. After fixing the hardware and making it actually work, the mag_hardware started to work as expected.
In fact I was unable to calibrate the AK8975 properly but that's another story (or maybe problems with the HW itself). Anyway it behaves the same way in nav and on cleanflight.

closed 2016-07-08 by krzysztofmatula

digitalentity, 2016-07-08:
I'm not sure it should behave like this. Imagine having a HMC5883 onboard and close to sources on interference (battery, buzzer) and AK8975 on the GPS module (selected explicitly using `mag_hardware`). 
If external compass fails for some reason INAV will still detect onboard one and will engage RTH/WP. But using an onboard compass **might result in fly-away** in described case!

digitalentity, 2016-07-08:
#336