iNavFlight/inav #302: FP-PID default values

DzikuVx, 2016-06-20:
Since many (most, for unknown reason) pilots tends to favor mwrewrite or even LuxFloat over FP-PID and many pilots complain that defaults are unflyable I've decided to compare them in terms of P response for error.

Since mwrewrite code is huge blob of unreadable mambo-jumbo, I compared it to LuxFloat after LF has been rescaled to match mwrewrite. So, we can have comparison of FP-PID to LF and mwrewrite.

To have common ground, common unit it `gyroADC * gyro.scale` I've called ***ADCg*** that if I remember correctly equals dps.

## FP-PID P-term response to error:

`Pterm = (ADCg_error * Pgain) / 40`

## FP-PID Pterm response on default P (30)

`Pterm = 0,75 * ADCg_error`

## LuxFloat P-term response to error:

`Pterm = (ADCg_error * Pgain) / 32`

## LuxFloat Pterm response on default P (40)

`Pterm = 1.25 * ADCg_error`

# Some conclusions

So, not only default LuxFloat P-term response on defaults is ATM much stronger, it also scales differently.
To archive the same response on FP-PID as LuxFloat has on P=40, FP-PID would have to have P=50. And FP-PID P=80 is equal LuxFloat P=64

I think that half of the problem with FP-PID are not tuned enough defaults, but half is psychological. Pilots are afraid to push P-gain high enough to have good results. They think in terms of mwrewrite where P=80 is a lot. While in world of FP-PID it is not that much after all. 

So, why not to do the same what CF and BF done with LuxFloat and not to rescale it CF defaults and use those defaults as FP-PID defaults? Of course this might be hard for D-term, but simple for P-term and pretty simple for I-term.

digitalentity, 2016-06-21:
I like the idea of unifying scaling with CF/BF. I did the math:

**PTerm:**
LUX: PTerm = 1 / 31.2195 * errorDPS * P8
FP:  PTerm = 1 / 40 * errorDPS * P8

**ITerm:**
LUX: ITerm += 1 / 4.092 * errorDPS * dT * I8
FP: ITerm += 1 / 10 * errorDPS * dT * I8

**DTerm**
LUX: DTerm = 1 / 1905,51688 * deltaDPS * pidProfile->D8[axis]
FP: DTerm = 1 / 4000 * deltaDPS * pidProfile->D8[axis]

Can you verify that I didn't do anything wrong and we'll try to unify scaling in 1.2

DzikuVx, 2016-06-21:
Pterm is correct, Dterm is plausible, it's hard to belive Iterm is ~400 times stronger in FP. 
I will run the numbers later

digitalentity, 2016-06-21:
It's not 4096 for ITerm, it's 4.096 = ~4. LUX is 2.5 times stronger than FP

DzikuVx, 2016-06-21:
OK, this I can belive. I will run the numbers after work.

martinbudden, 2016-06-21:
@DzikuVx , good work - I think it will be excellent to have the PIDs scaled to match Cleanflight/betaflight.

digitalentity, 2016-06-21:
Still, fine tuning will be different because of different algorithms and
filtering, but it will be close.

DzikuVx, 2016-06-21:
@digitalentity I checked your numbers and had the same value, so probably with #304 all controllers should behave similar on defaults. 
@digitalentity @martinbudden please check my numbers from PR, somehow I feel I got them the wrong way.

If numbers are correct, I will create a Wiki Page with conversion.
Next step would be to rescale LEVEL PID as well

digitalentity, 2016-06-21:
Yes, only LEVEL P please (LEVEL I is different from CF/BF)

DzikuVx, 2016-06-21:
Of course, only LEVEL P. I think I will create a change to configurator that will state that I is not I but LPF cutoff frequency

DzikuVx, 2016-06-21:
https://github.com/iNavFlight/inav/wiki/PID-conversion-from-pre-1.2-to-1.2

closed 2016-06-22 by digitalentity

Ralf-W, 2016-08-26:
@DzikuVx 
Hi Pawel,
we just found out that the wiki page is correct but there is a wrong formula (typo) in the spreadsheet at Google. https://docs.google.com/spreadsheets/d/133vfzz6_38W5nUmoRNuP7ZX9V1E-8IG6x0FxuxkBuQg/edit?pref=2&pli=1#gid=0
Field B12 must read B6/2,5 - but it does calculate B5/2,5. This is wrong.
Can you correct the spreadsheet or do you know who can do it?

DzikuVx, 2016-08-26:
sec...

DzikuVx, 2016-08-26:
@Ralf-W spreadsheet fixed, thanks for finding that

Ralf-W, 2016-08-26:
@DzikuVx 
Thanks!