iNavFlight/inav #305: NRF24 addition broken MSP_RX_CONFIG MSP message

digitalentity, 2016-06-21:
MSP_RX_CONFIG now have different format which messes up Configurator - see https://github.com/iNavFlight/inav-configurator/issues/22

digitalentity, 2016-06-21:
Introduced by #283

digitalentity, 2016-06-21:
@martinbudden We either need a new message (i.e. MSP_NRF24_CONFIG) or make changes to configurator to ensure it correctly processes MSP_RX_CONFIG and MSP_SET_RX_CONFIG for both INAV 1.1 and below and INAV 1.2 and above

martinbudden, 2016-06-21:
We need to change the configurator anyway, to allow the user to select which NRF24 protocol to use (since several are currently supported, and I plan to add more). I'm not in favour of adding a new message, since this is certainly falls within the category of RX configuration.

digitalentity, 2016-06-21:
@martinbudden then I suggest moving nrf24_protocol to the end of the message to minimize compatibility problems when using old configurator / CF Configurator. What do you think?

martinbudden, 2016-06-21:
I'm happy with that. I put it where it is for alignment reasons, but I think compatibility reasons are more important.

martinbudden, 2016-06-23:
Addressed in PR https://github.com/iNavFlight/inav/pull/309

Only briefly tested, so it would be good if someone else can confirm this works.

digitalentity, 2016-06-24:
I can confirm that it works. Thanks!

closed 2016-06-24 by digitalentity