iNavFlight/inav #310: Alternative way of arming.

oleost, 2016-06-23:
Because of lots of request of different ways of arming, or keeping it armed at all times...

Reasons: 

1: Aux arming needs and free channel, and many people have radios with only 6 channels and limited tx programming.
2: Don't want the aircraft to accidentaly disarm in flight..

Possible solution:

1: Arm with an switch connected to the RSSI connection.

2: Auto armed. (Example auto arm after steady GPS lock and no movement of aircraft for 5secounds.) Resets on battery cycle.

3: Always armed. Homepoint is as soon as throttle is applied.

4:New stick combination to arm/disarm. Current one is only one stick down to the side. Solution could be both left and right stick down and towards each other, for more than 3 secounds. Cant think of anyone holding those sticks that long in that position.

All noted solution is there of course requirred to have motor_stop so one can kill motors with throttle channel.

Also, there is the case of reboots of the FC in air. Could need some more security there, as an fixedwing is possible to fly passtrough mode and its not neccesary to have calibrated gyros etc.

Buzzliteyear, 2016-06-25:
Please dont be like the Arduplane guys and have our airplane mode always armed after arming and cant disarm
even though i fly a lot with my rudder id rather crash my plane than have this feature taken away
Is it possible to have the choice of a couple of ways what ever happens
no switches on airplane that seems archaic 
if anything the stick combo seems best as it doesnt take any extra channels 
maybe both stick low and to the right and left (arm-disarm) and only in Manual/Acro Mode
cheers

sppnk, 2016-06-25:
@oleost . I like the idea of some kind of auto-arm. For airplanes I dont figure out any situation where I should disarm in flight. Id est, if your plane fly-aways that is because you have no controls at all, so you wont be able to disarm if something goes wrong. 

The disarm by switching off the battery. Or something like that. I like the idea of have another channel for future uses.

oleost, 2016-07-15:
I'm currently having so I need both sticks to the right to disarm, and also hold them for 5s.

272b0b11e3b7bfb700779461996751ec62e7b742

Tried coding it so that it would only add delay if (feature(FEATURE_MOTOR_STOP).
But that was out of what I was capable of doing.

@digitalentity what do you think?

oleost, 2016-07-15:
Ehm, that code did not work as expected. It just hangs everything for 5s, and then disarm.. Need to code some more..

Buzzliteyear, 2016-07-16:
to loose height  I can use my full rudder and half my roll for sometimes at least 5 seconds thats why i have arm/disarm on a switch now, be great to connect them together so one doesnt disarm by misstake flicking the wrong switch, or at least have a choice in the CLI for either stick/switch stick/stick

oleost, 2016-07-16:
Switch arming won't be changed.

If you setup an aux on for arming it will only arm/disarm using that
switch.  No intention of changing that.

Yes,  I thought about that after, stick command will be changed to hold
them opposite way of each other.  So yaw left and roll right.

16. jul. 2016 1.17 p.m.

oleost, 2016-07-17:
@digitalentity 

Could you give me a hint why I can't get this delay to work?

c8b9fb320cabef04b8703c32dbc49fa54c6132c2

digitalentity, 2016-07-17:
@oleost 
because `rcDelayCommand` will probably always be < 1000000000

What are you trying to accomplish?

digitalentity, 2016-07-17:
Also, using this combination of stick arming will cause a lot of damage on multirotors.

oleost, 2016-07-17:
@digitalentity I'm trying to increase the minimum time the sticks need to stay in that position before disarming. 

Yeah,  the crocodile sign are the wrong way,  so I turned it around and lowered the value to 400, but that caused it to newer disarm.  I think probably the rcdelaycommand gets reset when it reaches 250.

So I tried making a new variable just for disarm and increment it the same way rcdelaycommand does.  But I didn't get it to work. 

Regarding multirotors if this was to be default I guess we need to make it only for fixed-wing and keep regular disarm for the rest. 

The issue now is that with my 4ch receiver on my airplane and only disarm on full rudder I'm guaranteed to disarm in air.

oleost, 2016-07-17:
Or I maybe just disable disarming for fixed-wing,  so it will require arming but not allowing disarm if used with motor_stop feature..  But for this to be implemented in inav it would be best that it was user configurable.  "disable_disarm_fixedwing"

Buzzliteyear, 2016-07-17:
Oleost what Tx system are you trying to use with 4 ch rx ?

oleost, 2016-07-18:
It's an Spektrum dx6i with an Ar400.

oleost, 2016-10-02:
New way of auto arming #375 

Closing this one.

closed 2016-10-02 by oleost

63Gio, 2022-03-21:
I am using INAV 3.0.1: is there any possibility to use a combination of commands not to waste a channel of the RC transmitter to arm a drone?
Thanks