iNavFlight/inav #319: Unify flaps handling

DzikuVx, 2016-06-24:
As an extension of #307 : unify flaps handling with single flight mode and config option: FLAPS, FLAPERONS, FLAPS + FLAPERON and allow to configure flaps throw with AUX channel

sppnk, 2016-06-25:
With this future feature and  with separated flaps, we could have 'airbrakes' (spoilerons + flaps). I would like to test that. 

In one of my planes I use separated flaps, and with a potenciometer I regulate the throws of the flaps (with a mixer customairplane setup and an AUX channel for the pot).

DzikuVx, 2016-06-25:
@sppnk this you can test right now. `flaperon_throw_inverted=ON` and FLAPERON activation on the same AUX that drives flaps

sppnk, 2016-06-25:
Oh, I didn't realize that. Thks