iNavFlight/inav #321: Force gyro_sync_denom=8 when gyro_lpf=256HZ

DzikuVx, 2016-06-26:
As a result of awesome investigation by @theArchLadder in #218 , force `gyro_sync_denom=8` when `gyro_lpf=256HZ`. This is because gyro enters 8kHz mode and higher denominator is required to keep looptime in reasonable ranges

theArchLadder, 2016-06-26:
Can F1 targets use looptime of 1000? If not, it may make more sense to multiply gyro_sync_denom by 8 when 256hz is used. That way `gyro_sync_denom=1` will give a looptime of 1000 and `gyro_sync_denom=2` will give a looptime of 2000, like it usually is...

DzikuVx, 2016-06-26:
Nope, F1 can not reach that low. Last time I checked it was about 1400us. But we can force gyro_sync_denom=16 depending on target

martinbudden, 2016-06-26:
The defaults in betaflight are:
```
    masterConfig.gyro_lpf = 0;                 // 256HZ default
#ifdef STM32F10X
    masterConfig.gyro_sync_denom = 8;
#else
    masterConfig.gyro_sync_denom = 4;
#endif
```
There is also msp code to set values according to looptime: https://github.com/betaflight/betaflight/blob/betaflight/src/main/io/serial_msp.c#L108-L174

DzikuVx, 2017-01-24:
With latest changes to Configurator, this ticket is obsolete. Configurator allows only to set correct values via GUI

closed 2017-01-24 by DzikuVx