iNavFlight/inav #322: It happens that ALT_HOLD makes strange things...

krzysztofmatula, 2016-06-27:
I observe that *sometimes* when enabling the ALT_HOLD the copter starts going up and rotates in the YAW axis relatively quickly. I don't yet have steps to reproduce the issue. It just sometimes happens. 

Before enabling BARO I don't do anything special. The craft is just hovering in HORIZON mode.

See the blacklog log attached at the following points:
- 7:01.886
- 7:24.554

The log reflects the real behavior - altitude increases and heading changes.

I use INAV 1.1

[blackbox_log_2016-06-27_214021.TXT](https://github.com/iNavFlight/inav/files/335942/blackbox_log_2016-06-27_214021.TXT)

DzikuVx, 2016-06-28:
Could you try similar scenario without MAG mode enabled? It looks like your magnetometer is too close to power lines and is affected by changing mag field generated by them. 

[image]

My theory is that when you enable ALT_HOLD and your throttle is far from hover value, sudden spike in current going into motors causes magnetometer going crazy and causing yaw correction.

Confirm by disabling MAG_HOLD or/and moving mag further away from power lines

DzikuVx, 2016-06-29:
> In the examples I gave the throttle was at a perfect hovering level.

Well, not here: 7:24.554. See what is happening with throttle right after you engage ALT_HOLD. There are quite big throttle movements. Maybe baro is not isulated enough? Sponge?

krzysztofmatula, 2016-06-29:
Baro sensor is insulated properly and usually works very well, giving ALT_HOLD with almost no oscillations. Here, just after changing mode the throttle applied is much over the hovering level and the quad goes up (what baro readings clearly show).

I bet it is the MAG issue, or some other bug in the code. I don't understand how the wrong MAG readings would end up in not keeping the altitude and going up.

DzikuVx, 2016-06-29:
Going up might be only a side issue here. Yaw jump is rapid and strong, that copter trying to archive rotation rate generates more thrust

krzysztofmatula, 2016-06-29:
That may be the explanation.

There's one more thing that *may* cause side effects. 
Switching on BARO with a transmitter switch actually is visible as a quick change over the following modes:
- HORIZON_MODE|MAG
- HORIZON_MODE|MAG|GPS_HOLD
- ANGLE_MODE|MAG|BARO|GPS_HOME
- ANGLE_MODE|MAG|BARO
- HORIZON_MODE|MAG|BARO

I use single AUX for providing BARO, GPS_HOLD and GPS_HOME switches multiplexed to the FC. It looks like someone smoothens the RC channel, which results in the above quick sequence. I've no idea how to disable such smoothing, and whether it on on the transmitter (er9x) or FC side. Nevertheless, I was always having this and it didn't influence ALT_HOLD.

krzysztofmatula, 2016-07-08:
I reproduced the issue with MAG disabled. See the attached log at 1:31.000.

I was hovering at altitude ~1.6m, then switched on ALT_HOLD. As a result the machine moved up quite rapidly and started to stabilize at altitude >9m.

It looks like it *wants* to go to such altitude, while I wanted it to stay at 1.6m.

I have the default nav_rth_altitude = 1000 and nav_rth_alt_mode = AT_LEAST - maybe the fact that RTH was active for a fraction of second while engaging ALT_HOLD caused what I observed? It looks more and more like a bug, especially that it happens "sometimes"...

[blackbox_log_2016-07-08_220647.TXT](https://github.com/iNavFlight/inav/files/355060/blackbox_log_2016-07-08_220647.TXT)

krzysztofmatula, 2016-07-08:
Confirmed!
Changing the nav_rth_altitude to 5m made the ALT_HOLD altitude close to 4m, and not 9m as before!
In the attached new log it occurs at 0:19.100.

Then, I programmed a dedicated switch for engaging ALT_HOLD (without going through RTH for a fraction of second) and - ALT_HOLD worked!. See it at 0:38.806.

Later I used the original switch again (the one that enables RTH for a fraction of second) to enable ALT_HOLD (at 0:53.804) and... it worked as it should. Well, didn't I say the issue manifests *sometimes* ?

[blackbox_log_2016-07-08_231556.TXT](https://github.com/iNavFlight/inav/files/355130/blackbox_log_2016-07-08_231556.TXT)

digitalentity, 2016-07-08:
@krzysztofmatula 
Ok, now that I know that you are switching to AltHold through (essentially _from_) RTH explains it. If altitude controller was active (RTH) switching to AltHold mode won't reset current altitude target. This is a bug that needs fixing. Thanks for finding.

closed 2016-07-08 by digitalentity

krzysztofmatula, 2016-07-09:
I've jus verified the fix and it solves the problem, but only partially.
My original observation was that the copter goes up *and rotates*.
After this fix the altitude is maintained, but it still rotates (unless I disable mag).
I think similar correction shall be done to HeadingController as well. It looks like it rotates towards the home point and stops. Again, surely result of short RTH.
I can upload a log if necessary, however I think it won't be needed.

krzysztofmatula, 2016-07-12:
Unfortunately I have no rights to reopen the issue...

digitalentity, 2016-07-12:
@krzysztofmatula current issue is resolved. Yaw rotation is another issue, please open a new one.

commit cc4c63ce (2016-07-08): Reset AltHold setpoint when transitioning from RTH/WP. Fixes #322