iNavFlight/inav #324: PPM does not work on  db7116b / NAZE32

stronnag, 2016-07-02:
PPM does not work on  db7116b / NAZE32. Probably as a result of the new IO work. 

PPM does work (flight tested) on F3 (Dodo) for db7116b.

Masheroff, 2016-07-02:
All builds before commit b5eda096a5a25cdedb53dc1b57caac0c36c1e873
Author: Martin Budden <mjbudden@gmail.com>
Date: Fri Jun 24 15:07:38 2016 +0100
Converted targets to use new IO

are working. So I think problem in this commit.

stronnag, 2016-07-02:
fyi @martinbudden

martinbudden, 2016-07-02:
OK, I'll have a look.

digitalentity, 2016-07-02:
@martinbudden I think `GPIO_MODE_FROM_IOMODE` macro is not working properly. It masks lower bits (0-1) while actual GPIO mode uses bits 2-6 on F1 CPUs. 
On F3 CPUs behavior is correct - GPIO mode uses bits 0-1 and macro works correctly.

I think PWM/PPM code should be reworked and `pwmGPIOConfig()` should be replaced with `IOConfigGPIO()`. I wonder if we shoud get rid of `gpioInit()` function alltogether.

martinbudden, 2016-07-03:
@digitalentity , fixed in PR https://github.com/iNavFlight/inav/pull/328

I agree, I think we should get rid of `gpioInit()` entirely. I wasn't around when it was introduced, but it looks like it was an attempt to simplify the IO and has now been superseded by the new IO code. There are very few instances of it left in betaflight and I think they will all eventually go.

digitalentity, 2016-07-03:
@martinbudden 
Thanks for fixing. How is migrating BF to new IO going?

martinbudden, 2016-07-03:
Most of BF has been migrated to the new IO. Most of the work has been done by @blckmn .

I'm going to start moving over IO code from betaflight now. I'll start with the LEDs and beeper, and work on from there. All of this IO works moves us closer to getting F4 working.

digitalentity, 2016-07-03:
@martinbudden thanks! Can't wait to see new IO and F4 support in INAV.

martinbudden, 2016-07-03:
You can follow my progress in PR https://github.com/iNavFlight/inav/issues/274 . I'm updating the task list as I go along. It says 8 out of 15 tasks complete, but then I add tasks as I progress...

digitalentity, 2016-07-03:
Fixed

closed 2016-07-03 by digitalentity