iNavFlight/inav #329: Decouple gyro update from PID loop

digitalentity, 2016-07-03:
Allow gyroscope to be sampled and filtered faster than PID loop makes corrections.

Settings:
`gyro_lpf` will define built-in filtering
`gyro_rate` will define gyro sampling rate. Possible settings 500HZ, 1KHZ, 2KHZ, 4KHZ, 8KHZ.
`gyro_sync` define if we allow gyro interrupt.
`gyro_pid_denom` - how much slower PID process will make corrections
`gyro_downsample_method` - what method will be used to downsample the gyro to PID frequency. Values: SCATTER, AVERAGE

This will make `looptime` deprecated.

Some hardware won't allow all possible settings, i.e. MPU6500 with gyro_lpf=256HZ doesn't respect the denominator and fires gyro_interrupt at fixed 8kHz rate.

digitalentity, 2016-07-03:
We can also chose to apply gyro low-pass filtering before or after downsampling. If we have CPU resources it's better to filter at highest possible rate, but on lower-performance boards filtering after downsampling is adequate enough.

DzikuVx, 2016-07-03:
I was looking at the code and yes, this is very good idea. I have some remarks to your idea:

1. `gyro_pid_denom` I think its obsolete to current looptime. If gyro and pid loops are decoupled, they can be fully decoupled: gyro at 1kHz, pid and 1.5kHz or 2.5kHz. Personally I'd like to have possibility to set whatever pid loop frequency I want to. Good UI for setting pid and gyro loops would be a must. This is starting to get complicated for non-developers
1. `gyro_downsample_method` is it really needed? If gyro task will have own LPF (like currently BiQuad) then pid loop would just take what is currently available on the filter output = current gyro value. Additional AVERAGE would act as secondary LPF introducing extra signal delay. And we do not like signal delay
1. I already mentioned it: gyro LPF would have to be applied in gyro loop. If not, I see no income from it. LPF sampling frequency should be as high as possible = in gyro loop. If LPF sampling frequency is down to PID loop, this is what we have right now
1. To save computing resources, I would think about also decoupling ACC readouts. ATM they are run in sequence. Do we really need ACC in 8kHz mode? With such low LPF cutoffs as currently, we can lower sample rate of ACC (comparing to gyro) to let's say 500Hz

martinbudden, 2016-07-03:
My comments:

1. `gyro_rate`, I think there is only a need for two values: 1kHz and 8kHz. 1kHz if internal gyro filtering is on, 8kHz if not. If minimal work is done on sampling I would expect all targets to be able to handle 1kHz. And if gyro filtering is off, then 8kHz sampling should be used.
2. `gyro_sync`, we should get rid of this setting and force gyro sync to be always on. I see no disadvantage in this if it is scheduled properly.
3. `gyro_pid_denom`, I don't think we should have this. Instead the pid should be split into its constituent parts - `pidGyroUpdate`, called after a gyro update; `pidRCupdate` called after new RC values received; and `pidCalculate` called just before the motors are updated. We might like to have a `motor_update_denom`, however.
4. `gyro_downsample_method`, perhaps a better name would be `gyro_filter_type` since downsampling is effectively applying a filter.
5. yes, `looptime` then becomes redundant.

digitalentity, 2016-07-03:
@martinbudden 
The idea is to avoid calculating full PID at 8khz rate - therefore `pidGyroUpdate` shoudn't calculate anything except gyroRate in dps otherwise F1 targets won't be able to handle it. `pidCalculate` will compute P, I and D and update motors. What do you think?

jflyper, 2016-07-03:
I don't know much about PID task, but does this change means gyro could be sampled asynchronously to (from?) PID task?

jflyper, 2016-07-03:
Hmm... I remember seeing something similar in the CF PR...

This one: cleanflight/#1858 ?

npsm, 2016-07-03:
After looking at some gyros datasheets I was thinking a similar approach, It would be beneficial to adjust the built-in analog filter of the gyro to the desired pid-rate, but sample at the highest possible frequency (with dma) this will help with noise reduction from both sensor and ADC. Probably not all hardware would allow this but this would result in best sensor accuracy.

digitalentity, 2016-11-21:
Done.

closed 2016-11-21 by digitalentity