iNavFlight/inav #331: Speeding up blackbox downloading

erpalma, 2016-07-03:
Hi guys, I'm a relatively new iNav/CF/drone user. Probably the only feature that I find irritating to use is blackbox downloading using the configurator. In order to solve this issue I suggest the following:

Use a higher baud rate for MSP (I successfully tested 921600 bps on SPRACING F3). I think that 115200 bps is WAY too low.

I also wrote a small python script to download the blackbox log using the CLI command flash_read (the target must be compiled with USE_FLASH_TOOLS). The [script](https://gist.github.com/erpalma/9f3e74b0f81384a558cb08e5e4c7b48e) should work also on windows, but I only tested on linux. It requires only one dependency: pyserial (pip install pyserial).
`$ python blackbox_dumper.py /dev/ttyUSB0 115200 blackbox_log.txt`
`Entering CLI...`
`Getting flash size...`
`Dumping...`
`[ ==                                                 ] 4.39% - 11.2 KB/s - ETA: 696s`

I can dump 8MB flash in about 12 minutes at 115200 and in about 90 seconds at 921600!

digitalentity, 2016-07-04:
Maybe just allow Configurator to connect at higher baud rates?

erpalma, 2016-07-04:
Yes sure, this is what I meant! I was in a rush while I wrote down the issue :)

I would keep 115200 as default but let the user configure MSP and connect to the FC with baud rates at least up to 921600. BTW I also tested the blackbox download through Configurator at 921600 bps and it took about 15 minutes if I remember correctly. Of course transferring over (CLI has no overhead)

oleost, 2016-07-20:
I tried adding more speeds in inav configurator, but I did not get it to work.

https://github.com/iNavFlight/inav-configurator/blob/master/main.html#L110

These are the speeds I tried to add

```
							<option value="256000 ">256000</option>
							<option value="460800 ">460800</option>
							<option value="921600 ">921600</option>
```

erpalma, 2016-07-28:
You have to set the same speed on the MSP port on the FC. See #365

DzikuVx, 2017-08-28:
Solved via  #2044 and iNavFlight/inav-configurator#229

closed 2017-08-28 by DzikuVx