iNavFlight/inav #332: G-tune in inav.

Jimpa86, 2016-07-03:
Have a spracing f3, running newest inav but can't find G-tune. I'm new to quadcopters and this is my first one.  How do I find it/enable it? 

Best regards

DzikuVx, 2016-07-03:
G-tune is not available in INAV at the moment

closed 2016-07-03 by DzikuVx

Jimpa86, 2016-07-03:
Ok, then I must have misunderstood. 
Reed these threads and thought it was there. 

https://github.com/iNavFlight/inav/blob/master/docs/Gtune.md

www.rcgroups.com/forums/showthread.php?t=2495732&page=85

DzikuVx, 2016-07-04:
If you want you can run gtune in cleanflight and then depending on inav version convert on just copy pid values

andre92, 2016-07-17:
How about bringing back G-tune for F3 targets? I understand it was removed to save space on F1 targets?

DzikuVx, 2016-07-17:
@digitalentity what do you think?

Linjieqiang, 2016-07-17:
I agree……