iNavFlight/inav #333: ALTHOLD mode is activated when the altitude falls problem!

meban, 2016-07-05:
Hi .. When I opened ALTHOLD mode, landing on the ground and floated quadcopter in place. Even if I did the maximum throttle, not rise. barometer completely insulation from light and the wind.There is very little vibration in the engines.

ZMR250
FC: Sp racing f3
Software : INAV 1.1
Mode: Angle-Mag-Nav Althold

Could you help ? What is the problem.

oleost, 2016-07-05:
Github is usually don't for support, that the forum thread for.

Anyway, if you read the wiki it would say a few things you should include when there's an issue. Else we are just playing bingo...

https://github.com/iNavFlight/inav/wiki/1.-Getting-started-with-iNav
Checklist if you're having issue with something:

Try and look through the wiki regarding the issue you have. You can also search the Wiki.
Read the first post rcgroups Cleanflight iNav thread. Also read the last 5 pages in the thread to see if someone else has already mentioned it. Also try and search in the thread.
Explain your issue, include CLI dump and blackbox log if you have a logger. Mention what you have tried, and also if it's working as intended in stock Cleanflight.

digitalentity, 2016-07-05:
@meban please consider using RCGroups for support. Dump and Blackbox log would be very helpful.

closed 2016-07-05 by digitalentity