iNavFlight/inav #336: Explicit mag_hardware is ignored if selected chip is not available

digitalentity, 2016-07-08:
If selected compass chip is not present, firmware will fallback to auto-detect any other chip which is not safe.

Imagine having a HMC5883 onboard and close to sources on interference (battery, buzzer) and AK8975 on the GPS module (selected explicitly using mag_hardware). 
If external compass fails for some reason INAV will still detect onboard one and will engage RTH/WP. But using an onboard compass might result in fly-away in described case!

krzysztofmatula, 2016-07-08:
I agree. Additionally the two mags may require different calibration and may have different orientation!

andre92, 2016-07-12:
I'm not sure this solves it but it should prevent the onboard mag from being used: https://github.com/iNavFlight/inav/pull/348

commit 49bfcd7b (2016-07-13): Update initialisation.c (#348)

Changed behaviour of mag_detect when forced sensor is not present
New behaviour: continue with mag disabled
Closes #336
closed 2016-07-13 by digitalentity