iNavFlight/inav #337: Release 1.2 preparations

digitalentity, 2016-07-09:
We have quite a lot issues open for 1.2 but I don't think they are all going to be closed soon. I think we need to reschedule some of them for 1.3. Ideas?

ricard0g0mes, 2016-07-09:
I think one big issue that should be resolved in 1.2 is the descend velocity in WP mode.
It just fall of the sky.

martinbudden, 2016-07-09:
It depends when you want to do the release.

As you know, the big one I am working on is issue https://github.com/iNavFlight/inav/issues/274 F4 target support. It will probably take me at least two weeks to finish that. I'm doing it incrementally so that at no stage is anything broken (at least not deliberately), so it doesn't have to be finished for release 1.2. 

Even if the full F4 port doesn't make it into v1.2, I would like the conversion to the new IO to be complete for v1.2 - that has been held up by a couple of bugs in the betaflight conversion, but those have been fixed now.

The other big one is issue https://github.com/iNavFlight/inav/pull/151 - migration to parameter groups. This could certainly be deferred to 1.3.

Looking through the other issues, there don't seem to be any that can't be deferred to 1.3

There are also some bugs, but, as far as I can tell, they all are bugs that also exist in 1.1, so in principle 1.2 could be released without fixing them. However there are probably a few you want to fix.

oleost, 2016-07-09:
Looking at the issues marked with 1.2 milestone I'm thinking you could delay all if them to 1.3 and simple release 1.2 as it is. Theres already alot of new changes compared to 1.1 and it would be nice with an "official" 1.2 release now.

oleost, 2016-07-09:
Maybe fix #300 if it's an easy fix

digitalentity, 2016-07-09:
@martinbudden I think #151 and #274 are big enough to consider a major change, so I'll retarget them to release 2.0. I'll hold the 1.2 until you finish migrating to new IO.

I'm in favor of trying to merge a few more things and release 1.2. We might even have 1.3 before we merge big stuff in.

martinbudden, 2016-07-09:
@digitalentity , that makes sense. I'm generally in favour of doing more frequent smaller releases.

oleost, 2016-07-15:
Would be nice with an Chrome release of iNav configurator at same time as 1.2 release. Especially because of the new rates.

digitalentity, 2016-08-30:
I'm declaring code freeze until 1.2. From now on until 1.2 is released only bugfixes will be merged.

digitalentity, 2016-09-06:
Ok, I'll wait for #561 and make a release.
We'll enter 1.2 with a few known bugs, but those bugs are probably local to certain F4 targets (which are experimental in 1.2 anyway)

digitalentity, 2016-09-07:
Great! We're done with everything scheduled for 1.2. Releasing today

closed 2016-09-07 by digitalentity