iNavFlight/inav #338: Convert device drivers to new IO

martinbudden, 2016-07-09:
- [ ] timers
- [ ] LEDs
- [ ] beeper
- [ ] inverter
- [ ] batter
- [ ] sonar
- [ ] uarts
- [ ] SPI
- [ ] I2C

digitalentity, 2016-07-22:
@martinbudden I believe ADC is done?

martinbudden, 2016-07-22:
Yes, it is. And the checkbox above is marked.

digitalentity, 2016-07-22:
Yes, I marked it today after fixing ADC for FURYF3 target.

digitalentity, 2016-07-23:
@martinbudden I see you rescheduled this to 1.3, do you think it's wise to make a release while new IO is half-way?

martinbudden, 2016-07-23:
I must have done that by accident. I agree, this should be finished for 1.2.

closed 2016-08-08 by martinbudden

digitalentity, 2016-08-08:
@martinbudden thanks and well done!

martinbudden, 2016-08-08:
Thanks. I'm pleased with the end result. Some useful tidying of the drivers and convergence with betaflight happened along the way.

It should fairly straightforward to get an F4 target going now. I'll look into that in the next day or so.

If there are any new F3 targets you want to add in, you may as well do them now.

oleost, 2016-08-08:
Not sure if you guys want to do it,  but it's been a couple of requests on the forum to get the f3fc from rcexplorer.se working. 

A pull request over at 
Cleanflight https://github.com/cleanflight/cleanflight/pull/2283

martinbudden, 2016-08-08:
Yes, now this work is complete we can start adding new targets. If @lkaino submits a PR, we can include it.

lkaino, 2016-08-08:
Well I made the target on weekend, but it might require some small changes now for the new drivers.

I also need to include the barometer support for the integrated tricopter board. Problem is that I don't have a working tricopter board to test on currently. Will ask for testing help on rcexplorer.se forums.

martinbudden, 2016-08-08:
OK, make a PR whenever you are ready. You might also like to make a PR for betaflight.

I've just had a look at your site, and your baby tricopter is a beautiful machine!

lkaino, 2016-08-08:
The rcexplorer.se site and the copter there are David Windestål's making :). I'm just helping with the Triflight fork of Cleanflight.

I'll rebase my commit.

The Betaflight target has been ready for quite a while, Rotor Riot guys were supposed to test the hex with the F3FC racing board but haven't yet reported back yet.

@martinbudden I remember you once wrote instructions on how to make a pull request to a different fork of Cleanflight. Any chance you remember where you wrote it? :)

oleost, 2016-08-08:
@ikaino Guess the chance are high that this guy want to test it for you.  http://www.rcgroups.com/forums/showthread.php?t=2495732&page=504#post35438084

martinbudden, 2016-08-08:
@Ikaino , I'm not sure where those instructions are. I'll try and find them.

wilco1967, 2016-08-08:
F3FC would be great !

lkaino, 2016-08-08:
@martinbudden I think it was a comment you made in one of betaflight issues a while back :).

Here's the HEX, I will post it to the RCG thread and in rcexplorer.se forums for testing.

[inav_1.2.0_RCEXPLORERF3.zip](https://github.com/iNavFlight/inav/files/407389/inav_1.2.0_RCEXPLORERF3.zip)

lkaino, 2016-08-08:
I added a new issue for testing the target here: #438 

Let's not pollute this issue with the matter anymore :).