iNavFlight/inav #339: Help with Code or CLI for having a Failsafe Mode to turn Feature Failsafe on and off

DaleKramer, 2016-07-09:
I need to add a Mode on the Configurator Modes tab that will allow me to turn the Feature Failsafe on and off with any channel.

I have had a look at the current master and i get lost quickly.

Could someone give me a basic road map of what I have to do in code so I can tackle this?

Or is there a way to CLI a mode that will do this?

I want to turn Feature Failsafe OFF when throttle is below 1050 and ON when above.

krzysztofmatula, 2016-07-09:
I think you can't disable features by configuring modes.
The Failsafe mode as seen in the configurator allows to manually trigger the failsafe action with an AUX. However on inav it worked differently than real failsafe triggered by RX for me (AFAIR it didn't perform RTH, while RX triggered failsafe does).

DaleKramer, 2016-07-09:
It looks like a relatively involved procedure (maybe not if you know what you are doing :) to accomplish what I want to do in a general way and have the ability to chose to use/or not use this new mode or feature, or whatever it should be called.

But, I think this one liner in my personal code branch would work:

```
static void failsafeActivate(void)
{
    if (rcData[THROTTLE] < 1050 ) return;  //dale prevent entering failsafe stage 2 when throttle below 1050
    
    failsafeState.active = true;
    failsafeState.phase = FAILSAFE_LANDING;
    ENABLE_FLIGHT_MODE(FAILSAFE_MODE);
    failsafeState.landingShouldBeFinishedAt = millis() + failsafeConfig->failsafe_off_delay * MILLIS_PER_TENTH_SECOND;

    failsafeState.events++;
}
```
I know this is NOT a way to be able to choose whether I want to prevent stage 2 Failsafe with a transmitter switch OR even be able to ever stop using my one liner but I am OK with this being in there all the time for me.

I think it still lets stage 2 failsafe activations that began when the throttle was above 1049, continue to completion or be cancelled by return of signal etc. 

Could someone tell me if this is the best place to put this kludge?

DaleKramer, 2016-07-29:
Unless a better option appears, I will be using my above code.  I use a throttle cut switch on my transmitter that lets me turn off and on, my single Spektrum sat/FC controlled electric motors in the air so I can fly my craft around in aircraft mode on its gas engine and high diversity RX/servos.  I do not want my electric motors doing a failesafe RTH when I am in throttle cut mode.

gaelj, 2016-07-29:
This seems sensible to me.
What if I add a CLI command called `disable_failsafe_on_throttle_cut`, which would need to be set to `ON` in order to activate your modification ?
I will issue a PR.

gaelj, 2016-07-29:
@DaleKramer Could you please test the PR and give feedback.

DaleKramer, 2016-07-29:
@gaelj Thanks, that looks great!  I am off on a 3 week trip but I will gladly test it when I get back.

danarrib, 2017-02-24:
If I understood, this new variable `disable_failsafe_on_throttle_cut` makes FAILSAFE not kick in if throttle value bellow 1050.

So if one land the aircraft, then turn the radio transmitter OFF before disarm the aircraft, it'll not start failsafe procedure.

This is indeed a nice feature. Just need to be well documented. I can see some situations when would be nice to have RTH even when throttle value < 1050... Think some glider going out of radio range, or a misconfigured RX Failsafe procedure, that set specific throttle value when transmitter is out of range.

oleost, 2017-03-28:
#1475

closed 2018-05-27 by stale[bot]