iNavFlight/inav #344: Disable min throttle in airline fixed wing

oleost, 2016-07-12:
With current code we need airmode to get full stabilization with zero throttle applied. 

But with airmode enabled the motor only goes down to min throttle and will keep  spinning even with throttle stick at zero.  So one needs to  lower the min throttle  on fixed wing the way it is now.

digitalentity, 2016-07-12:
Or maybe re-allow MOTOR_STOP when AIR_MODE is enabled.

oleost, 2016-07-12:
Yes that would be the correct way probably.

digitalentity, 2016-07-12:
Actually, this is a bug from the past. AIRMODE was adapted for fixed wings with MOTOR_STOP, but the MOTOR_STOP override was never removed.

closed 2016-07-12 by digitalentity

oleost, 2016-07-12:
That was quick :+1: 

Thank you, will update wiki to remove to need for lowering min throttle.

oleost, 2016-07-13:
@digitalentity doesn't not  your fix break airmode with multirotors?

digitalentity, 2016-07-13:
@oleost no, it doesn't AIRMODE is still active on mutirotors. Moreover enabling AIRMODE now doesn't break MOTOR_STOP - if you have MOTOR STOP your motors will stop at zero throttle regardless of airmode.

oleost, 2016-07-13:
@digitalentity OK. 

I thought one of the ideas with airmode was that  one could still use motor_stop,  but when flying  acrobatic and fast decent one could fly with airmode and override motor stop.  Since airmode is a flight mode you can change in while flying

digitalentity, 2016-07-13:
I think that since MOTOR_STOP is a feature, not a mode it shouldn't be overriden. We might want to make it a flight mode though, or create a special mode to override it (i.e. IDLE_UP) if necessary.

oleost, 2016-07-13:
Originally  airmode was actually called idle_up.

digitalentity, 2016-07-13:
Yes, I remember it but it was actually a hack to allow I-term on zero stick throttle (that time even disabling MOTOR_STOP didn't give you full control at zero throttle) 
Airmode it's now smarter than that, so there's no need for it to override MOTOR_STOP.

oleost, 2016-07-13:
With the way airmode works now is there a reason for it being a flight mode instead of just always enabled? 

Because now there are anti windup as well?  So even staying at ground with airmode now doesn't  do anything bad?

digitalentity, 2016-07-13:
It might cause spool up after landing. I'm flying with it always enabled and it doesn't do anything bad for me. Maybe make it a feature?

oleost, 2016-07-13:
I'm always flying with it enabled,  but then again I don't let my quad just stay on ground spooled up either. 

Reading Cleanflight notes it says "Note that AIRMODE will also overrule motor stop function! It will basically also act as an idle up switch"  https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md

csurf, 2016-07-20:
I just commented on issue [#350](https://github.com/iNavFlight/inav/issues/350)... This whole AIRMODE thing is simply wrong & broken IMO for fixed-wing. We should do away with it altogether for FW, and simply have full-scale PID corrections at all times: low throttle, zero throttle, and even when disarmed. I haven't looked deeply into the code to figure out what type of PID scaling is going on that's tied to the throttle value, but I think we should simply add a branch in the logic that forces full scaling for PID corrections for FW, and at *all times*, not only when armed & above min_check.

digitalentity, 2016-07-21:
In latest code with AIR MODE enabled you have full PID output even with zero throttle. I don't see an issue in assigning AIR MODE to any unused AUX and have it always enabled on airplane.

commit 16e21dfe (2016-07-12): Don't allow AIRMODE to override MOTOR_STOP. Closes #344