iNavFlight/inav #345: Transition from RTH to ALTHOLD makes the copter rotating towards home point

krzysztofmatula, 2016-07-12:
This is continuation of issue #322.

In my current transmitter setup enabling ALT_HOLD also enables RTH for a fraction of second. It is too short to actually perform any RTH activities, however when the ALT_HOLD is eventually engaged, the aircraft keeps rotating in YAW axis towards the home point.

andre92, 2016-07-12:
RTH updates setpoints for althold and maghold. As long as your throttle and yaw are centered, althold and maghold will not be updated again. 
Throttle and yaw input should solve this for you. 
to entirely fix this issue, RTH behaviour could be changed to set althold to current alt and maghold to current heading when disabling it.

krzysztofmatula, 2016-07-12:
Yaw input stops the rotation.
Is't actually minor issue for me, however nice to be fixed :)
The altitude has been already fixed, as described in #322.