iNavFlight/inav #347: iNAV 1.1: Wrong Blackbox data for tail servo

Ralf-W, 2016-07-12:
@DzikuVx 
I try to tune the PID of my tricopter using blackbox log and found that the values of the signal to the tail servo were written incorrectly into the log: I expect to find values reaching from 1000 (full left), 1500 (center) to 2000(full right). However I do find values ranging from 1500 to 2500. 

I assume the problem to be with the sign of the values in blackbox.c in line 

608 blackboxWriteSignedVB(blackboxCurrent->servo[5] - 1500);

But my understanding of C programs is not sufficient to kill the bug.

Here is a picture from the BB log of a test on the bench (without propellers). At time 00:01:100 the servo signal should become smaller than 1500. However it is transformed into a staircase trace up to 2500.
[image]
[iNAV_YAW.TXT](https://github.com/iNavFlight/inav/files/359278/iNAV_YAW.TXT)

DzikuVx, 2016-07-12:
I will try to take a look at it soon, since I will be somehow going into blackbox

Ralf-W, 2016-07-12:
Thank you. 
I really like your blog about PID, the excellent examples, and your explanations given on www.quadmeup.com

digitalentity, 2016-08-29:
Is the issue stil present?

DzikuVx, 2016-08-29:
Probably, I did nothing here after all. Leave it open, I will try to investigate

digitalentity, 2017-04-04:
I don't have a tri to verify. @stronnag can you check if the issue is still present?

Ralf-W, 2017-04-04:
Though I am not @stronnag here is a fresh log from iNAV 1.5.1 in my tricopter. (Sorry about the noisy motor signals. Meanwhile I balanced the props properly.)

[blackbox tail](https://cloud.githubusercontent.com/assets/16436064/24670880/0e6b9e84-1970-11e7-9a74-0a9a1f93a7cf.jpg)

[Heck im POSHOLD.TXT](https://github.com/iNavFlight/inav/files/893968/Heck.im.POSHOLD.TXT)

stronnag, 2017-04-04:
I think this bug is still present; here's my tricopter log summary (Version / Min / Max) since forever. Not sure it makes any sense, other than very few readings below c. 1450. 

http://seyrsnys.myzen.co.uk/inav_tail/inav_tail.ods

digitalentity, 2017-04-05:
Fixed and tested

closed 2017-04-05 by digitalentity

stronnag, 2017-04-06:
Confirmed (if you want a second opinion:)
````
Git Version	Tail_Min	Tail Max
INAV 1.6.3 (722f377f) SPRACINGF3EVO	1441	1723
INAV 1.6.3 (722f377f) SPRACINGF3EVO	1359	1804
INAV 1.6.3 (722f377f) SPRACINGF3EVO	1492	1578
INAV 1.6.3 (722f377f) SPRACINGF3EVO	1450	1707
INAV 1.6.3 (722f377f) SPRACINGF3EVO	1375	1629

````