iNavFlight/inav #349: MAG shows up connected all the time when using mag_hardware = 4

oleost, 2016-07-12:
When using it with an DJI naza gps and "mag_hardware = 4" the mag show that its connected all the time.

I can physically disconnect the GPS, and even after a reboot of FC it still shows that MAG is connected.

Can confirm issue on both 1.1 and 1.2. And on both Naza32 and spracingf3 with mag physically removed.

digitalentity, 2016-07-13:
Yes, this bug is known. The problem is that mag is only detected on boot and GPS is detected after that so if using NAZA GPS we don't really know if GPS/MAG is connected at boot and MAG is assumed to be present.

digitalentity, 2016-12-19:
With hardware failure detection in place we can actually address this bug

digitalentity, 2017-01-28:
I figure this will be handled by hardware failure detection and unconnected compass will be shown as failing.

I'm buying myself a genuine DJI GPS unit to verify/test/improve support for DJI hardware.

oleost, 2017-01-28:
@digitalentity Just tested with NAZA GPS.

Unplugging the GPS it will show up as GPS=UNAVAIABLE with red icon on configurator. Cloosing this one as this is fixed now.

closed 2017-01-28 by oleost

digitalentity, 2017-01-29:
@oleost thanks for testing!