iNavFlight/inav #350: Discussion - Convert AIR_MODE to a feature

digitalentity, 2016-07-13:
Continuation of #344.
As AIR_MODE is in general required to have better flight performance we should consider changing it to a feature that can be enabled and be always active. Might also rename to "LOW THROTTLE STABILITY". 
Thoughs?

DzikuVx, 2016-07-13:
Personally I am against it. I find it useful to disable AIR MODE in some cases while in flight. I do it during landing preparations on my 600m quad.

martinbudden, 2016-07-13:
I also like the ability to switch AIR_MODE on and off.

oleost, 2016-07-13:
Maybe name change?

Since if it's not overriding motor stop,  then it's not the same flight
mode as the orginal airmode anymore.

digitalentity, 2016-07-13:
Maybe split it into two:
AIR MODE for better flight stability and IDLE UP for overriding MOTOR_STOP. What do you think?

sppnk, 2016-07-13:
Personally, I dont have experience with iNav in multicopters, but for fixed wing, if it is a feature to improve flight, I guess it should be always active, and not configurable anyway.

kronosol, 2016-07-13:
Airmode can cause crazy behavior of your quad if its enabled while landing, at least it´s not recommend. 
So you will need to disable it bevor the quad touches the ground, i have airmode on the same switch as landing gear, so in general im safe.

stronnag, 2016-07-14:
AIRMODE is already a feature in betaflight (3.0 at least). I always have the mode enabled with iNav, with no adverse results.

csurf, 2016-07-20:
For FW, so-called "AIRMODE" (full-force PID corrections at low throttle) should be considered necessary functionality that's integrated into FC logic. FW aircraft need full PID corrections at *all times*, even when plane is disarmed. Throttle position should have no effect on the amount of PID corrections, except in the special case of TPA, where the user is consciously limiting PID correction at high throttle. However, at low or zero throttle (or even when disarmed), the servos should always correct with full force. The only time PID corrections should be forcibly disabled is when in PASSTHRU mode. This is  pretty much the standard way of doing it in other FC firmwares, but unfortunately it's currently broken in iNav...

oleost, 2016-07-20:
If you want airmode on at all time simple assign it to an unused aux
channel and leave it activate all the time.

I really like to enable airmode same time as armed, so I gain won't grow
before I'm ready to takeoff.

20. jul. 2016 12.41 p.m.

closed 2016-11-21 by digitalentity