iNavFlight/inav #351: nav_user_control_mode ATTI sucks, CRUISE is unsafe, how about blending them?

theArchLadder, 2016-07-14:
`nav_user_control_mode = ATTI` gives manual override when the stick is not centered, that's safe and a good default in case anything goes wrong with compass or gps reading. But it sucks to fly with, it's jerky and just doesn't feel right.

`nav_user_control_mode = CRUISE` feels like it should and is awesome to fly in, but one can get a fly-away if compass/gps reading is wrong, so it's not as safe and not a good default.

How about blending them? If stick is <75%, use CRUISE, if stick is close to full roll/pitch use ATTI, that way it should feel awesome around center stick but still allow manual override at full stick deflection if needed.

Thoughts?

kronosol, 2016-07-15:
Mmh,.. So what you mean is like the horizon mode in comparison to the angle mode.
I had something like the same idea while testing your tipp in #280,.. now you linked me here, i did some serious brain storming,.. :D. I came to the end, that it´s the right way you say it.
But I see one problem:
I imagine the behavior of the quad. The quad has a flyaway on pitch in front direction, it´s not level.
Inav will set the angle to the max which is default 30°.
Now you try to rescue it with a stick command in the other direction. Now on the "breaking" point as you say 75% off center the Quad does a hard movement from the 30° in front direction to like 20° on back direction. This cause a crazy shaky thing as when you go back under 75%, Inav will go back to the 30° in front direction.
I would prefer something like this:
On the "breaking" point, the quad first go to a level mode, so it doesn´t accelerate anymore.
Than on 80% of center it starts doing this override stuff as you say.
This will make the hole thing a little bit more smooth.
In the end, the only thing what really helps is to turn off PosHold,
so all we can do here is getting some more time in a panic moment ;).

digitalentity, 2016-07-15:
I think this feature (CRUISE/ATTI automatic transition in flight) would cause a very unpredictable behavior. Let's say I have max_speed set to 2 m/s and I'm trying to fly along with the wind that's 5m/s strong in CRUISE mode. At any requested **forward** velocoity (forward pitch stick) copter will actually pitch **backwards** to fight the wind. Then all of a sudden I hit the 75% stick and copter will instantly pitch forward and will accelerate like crazy.

I think about making a CRUISE/ATTI a modifier for POSHOLD like I did with NAV GCS mode. It will be a mode that you can switch from your TX but the only thing it will do is modify the POSHOLD behavior.

DzikuVx, 2016-07-15:
I'm rather against blending ATTI and CRUISE. Transition from one to second would be a sudden change of behavior. This is the same reason I never fly Horizon. 
But I like the idea of CRUISE/ATTI a flight mode

skaman82, 2016-07-17:
I still didn't get why CRUISE is unfase. I just tested it tody and liked it much better then ATTI

kronosol, 2016-07-17:
@skaman82 Because on Cruise mode you can´t override the flight controller in case of a GPS Glitch, you first have to disable PosHold. 
If you get a GPS Glitch on ATTI it will also do a movement but you can override it just with your stick command. In case of a panic moment you might get some more time before crash.

gaelj, 2016-07-29:
Can't you solve the issue by disabling PosHold in case of flyaway ? Flipping an RC switch doesn't take long, albeit slightly longer than pulling on your sticks.

oleost, 2016-07-29:
@gaelj My tought aswell. I have had issue with GPS due to user error in the beginning. Simple switching out of Poshold was the natural thing to do. Not try and fight it with the sticks.

gaelj, 2016-07-29:
OK, closing.

closed 2016-07-29 by gaelj