iNavFlight/inav #352: How far is iNav behind Cleanflight?

dantheman213, 2016-07-14:
Hey guys! Sorry was just curious, what's the turn around time for patching changes from the official Cleanflight repo into this one? Or do you not do that? Just wanted to understand your development paradigm before beginning to flash my fligiht controllers with this version of firmware. Thanks!

krzysztofmatula, 2016-07-15:
Other question is how much ahead it is :)

digitalentity, 2016-07-15:
In general INAV is behind CF in two things - internal configuration storage and eye-candy features (LED Strip), the rest is pretty much ahead.

closed 2016-07-15 by digitalentity

dantheman213, 2016-07-16:
Just to clarify --do you regularly patch in updates that the official Cleanflight devs/repo publishes? Or is this a complete fork and separation from Cleanflight?

digitalentity, 2016-07-16:
@dantheman213 TNAV and Cleanflight have diverged too much, it's not possible to patch updates from CF without much effort. However there is work being done on aligning INAV with Betaflight in low-level codebase.

dantheman213, 2016-07-16:
Is your goal ultimately to usurp Cleanflight's position or will you eventually merge codebases back in with Cleanflight? Thanks for answering my questions!

digitalentity, 2016-07-16:
My goal it to fill the apparently empty niche of affordable navigation-capable flight control. If this will and up merging things back to CF - excellent, if not - we'll manage :smiley: