iNavFlight/inav #354: Confused about hint for INAV 1.2

Ralf-W, 2016-07-15:
@oleost  In https://github.com/iNavFlight/inav/blob/master/README.md I find 
"Starting at 22-06-2016 INAV uses the same scaling for PIDs as Cleanflight/Betaflight LuxFloat and MWRewrite PID controllers. That means the following:
•PIDs from CF/BF can be used in INAV, no need to retune for INAV
•INAV uses the same PID defaults that Cleanflight and Betaflight
•Current INAV tunes can be converted to new using this guide. This applies to all INAV 1.1"

What does this mean for a "normal" user who has installed INAV 1.1 (May 2016).

To my understanding, the text in the readme already applies to upgrading from INAV 1.1 to INAV 1.2, which will need these changes of PID.

If my assumption is correct, I propose this text:

Changing from different firmware (e.g. Cleanflight/Betaflight) to INAV:
At the moment the stable release of INAV is version 1.1. It uses a different scaling for PIDs than Cleanflight/Betaflight LuxFloat and MWRewrite PID controllers. That means the following:

•Current CF/BF tunes can be converted to INAV 1.1 using this guide."

With the release of INAV 1.2 this will be changed:

•PIDs from CF/BF can be used in INAV 1.2, no need to retune for INAV
•INAV 1.2 uses the same PID defaults as Cleanflight and Betaflight
•INAV 1.1 tunes can be converted to new using this guide. "

And for the time being, the guide should be changed accordingly:

"With INAV 1.1, PID scaling and default differ from Cleanflight/Betaflight LuxFloat and MWRewrite PID controllers.

When migrating from CF/BF to INAV 1.1, use following conversion:

ROLL/PICH/YAW

 INAV 1.1 P_gain = CF/BF_P_gain * 1.3 

 INAV 1.1 I_gain = CF/BF_I_gain * 2.5 

 INAV 1.1 D_gain = CF/BF_D_gain * 2.1 

LEVEL

 INAV 1.1 LEVEL P = CF/BF_LEVEL_P * 6.1 

With INAV 1.2, PID scaling and default values change to match Cleanflight/Betaflight LuxFloat and MWRewrite PID controllers.

When migrating from INAV 1.1 to INAV 1.2, use following conversion:

ROLL/PICH/YAW

 INAV 1.2 P_gain = INAV 1.1_P_gain / 1.3 

 INAV 1.2 I_gain = INAV 1.1_I_gain / 2.5 

 INAV 1.2 D_gain = INAV 1.1_D_gain / 2.1 

LEVEL

 INAV 1.2 LEVEL P = INAV 1.1_LEVEL_P / 6.1

oleost, 2016-07-15:
Hi i didn't change that part of the readme.txt.

oleost, 2016-07-15:
Hi i didn't change that part of the readme.txt. 

But I would just change it to :

Starting at iNav 1.2 release(and current master).....

closed 2016-07-20 by DzikuVx

Ralf-W, 2016-07-20:
@DzikuVx 

Thank you!

I still have to learn how to edit / change such a page - to me it seems to be very different from changing the wiki.