iNavFlight/inav #359: FrSky Telemetry (softserial) broken?!

leramleram, 2016-07-18:
Hello

Given situation: 
Flip32+ flashed with INAV 1.1 (last stable release).
Ublox M8N GPS connected at 19200Baud on softserial 2.
Serial RX (SBUS) connected at 19200Baud on UART1.
Telemetry connected on softserial 1.

Somehow i can´t get FrSky telemetry to output any data on the selected port when using INAV. (I looked via oscilloscope and nothing happens on that pin.)
This was working before in Cleanflight, but to double check, I reflashed Cleanflight.
In cleanflight (with the same settings) telemetry is working as it should.

Can you help me pointing out what´s going wrong?

regards
Christian

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/NAZE 1.1.0 May 11 2016 / 02:37:17 (1f49891)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature VBAT
feature RX_SERIAL
feature SERVO_TILT
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map TAER1234

# serial
serial 0 1 115200 38400 0 115200
serial 1 64 115200 38400 0 115200
serial 30 4 115200 38400 19200 115200
serial 31 2 115200 19200 0 115200

# led
led 0 0,0:::0
led 1 0,0:::0
led 2 0,0:::0
led 3 0,0:::0
led 4 0,0:::0
led 5 0,0:::0
led 6 0,0:::0
led 7 0,0:::0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 2000
set emf_avoidance = OFF
set i2c_overclock = OFF
set gyro_sync = OFF
set gyro_sync_denom = 2
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 16
set rssi_scale = 30
set rssi_ppm_invert = OFF
set rc_smoothing = ON
set input_filtering_mode = OFF
set min_throttle = 1040
set max_throttle = 1980
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 1000
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set reboot_character = 82
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_dyn_model = AIR_1G
set gps_auto_config = ON
set gps_auto_baud = ON
set inav_accz_unarmedcal = ON
set inav_use_gps_velned = OFF
set inav_gps_delay = 200
set inav_gps_min_sats = 6
set inav_w_z_baro_p =  0.350
set inav_w_z_gps_p =  0.200
set inav_w_z_gps_v =  0.200
set inav_w_xy_gps_p =  1.000
set inav_w_xy_gps_v =  2.000
set inav_w_z_res_v =  0.500
set inav_w_xy_res_v =  0.500
set inav_w_acc_bias =  0.010
set inav_max_eph_epv =  1000.000
set inav_baro_epv =  100.000
set nav_disarm_on_landing = OFF
set nav_use_midthr_for_althold = ON
set nav_extra_arming_safety = ON
set nav_user_control_mode = ATTI
set nav_position_timeout = 5
set nav_wp_radius = 100
set nav_max_speed = 300
set nav_manual_speed = 500
set nav_manual_climb_rate = 200
set nav_landing_speed = 200
set nav_land_slowdown_minalt = 500
set nav_land_slowdown_maxalt = 2000
set nav_emerg_landing_speed = 500
set nav_min_rth_distance = 500
set nav_rth_tail_first = OFF
set nav_rth_alt_mode = AT_LEAST
set nav_rth_altitude = 1000
set nav_mc_bank_angle = 30
set nav_mc_hover_thr = 1500
set nav_mc_min_fly_thr = 1200
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_bank_angle = 20
set nav_fw_climb_angle = 20
set nav_fw_dive_angle = 15
set nav_fw_pitch2thr = 10
set nav_fw_roll2pitch = 75
set nav_fw_loiter_radius = 5000
set serialrx_provider = SBUS
set spektrum_sat_bind = 0
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 2
set align_board_yaw = 0
set gyro_lpf = 42HZ
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 50
set imu_dcm_kp_mag = 10000
set imu_dcm_ki_mag = 0
set pos_hold_deadband = 20
set alt_hold_deadband = 50
set yaw_control_direction = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1050
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = RTH
set rx_min_usec = 885
set rx_max_usec = 2115
set acc_hardware = 0
set baro_use_median_filter = ON
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set magzero_x = -187
set magzero_y = -89
set magzero_z = -93
set acczero_x = 0
set acczero_y = 0
set acczero_z = 0
set accgain_x = 4096
set accgain_y = 4096
set accgain_z = 4096

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 h
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

# aux
aux 0 0 5 1300 2100
aux 1 1 2 1200 1700
aux 2 2 2 1700 2100
aux 3 25 8 1300 1700
aux 4 7 4 1300 2100
aux 5 3 10 1300 2100
aux 6 10 3 1700 2100
aux 7 9 6 1300 2100
aux 8 12 9 900 1300
aux 9 23 5 1700 2100
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 868 2000 1500 90 90 53 5
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set nav_alt_p = 155
set nav_alt_i = 60
set nav_alt_d = 0
set nav_vel_p = 180
set nav_vel_i = 50
set nav_vel_d = 5
set nav_pos_p = 65
set nav_pos_i = 120
set nav_pos_d = 0
set nav_posr_p = 180
set nav_posr_i = 15
set nav_posr_d = 100
set nav_navr_p = 10
set nav_navr_i = 5
set nav_navr_d = 8
set deadband = 5
set yaw_deadband = 5
set throttle_tilt_comp_str = 0
set default_rate_profile = 0
set gimbal_mode = NORMAL
set mag_declination = 0
set p_pitch = 30
set i_pitch = 16
set d_pitch = 56
set p_roll = 30
set i_roll = 16
set d_roll = 56
set p_yaw = 100
set i_yaw = 40
set d_yaw = 0
set p_level = 120
set i_level = 15
set d_level = 75
set max_angle_inclination_rll = 300
set max_angle_inclination_pit = 300
set gyro_soft_lpf_hz = 60
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 40
set tpa_rate = 0
set tpa_breakpoint = 1500

#

oleost, 2016-07-18:
You forgot to read the release notes.

leramleram, 2016-07-18:
To be honest I´ve been looking, reading, and trying for hours already.
Seems I´m blind, and I might have overseen it, but would you be so kind and tell me and maybe others that possibly run into the same situation, what the solution for my problem is?

regards
christian

DzikuVx, 2016-07-18:
@leramleram INAV by default does not provides other telemetry protocols than LTM on STM32F1 targets, including Flip32. This is due to limited flash. You might want to compile own version with SmartPort enabled. Or drop me an email, I will compile you on pspychalski@gmail.com

closed 2016-07-18 by DzikuVx

sppnk, 2016-07-19:
IIRC, althought you can enable Frsky for F1 targets compiling your own version (as I do), you cannot use softserial (see #177).

What I do with my nazes and cc3d FCs is to share the Frsky telemetry with MSP in USART1. It works fine.

oleost, 2016-07-19:
He is using Flip32/Naze32. No issue in using Frsky telemetry over
softserial.

AABatteries, 2016-10-29:
Is it possible to recompile newest release for Naze32 with SmartPort telemetry enabled? Will there be enough flash memory on the chip? I wish I knew how to do it myself, and this is the only feature I'm missing from iNav now. Is there any help available on how to compile firmware yourself/enabling features? Thank you all in advance

DzikuVx, 2016-10-29:
You would have to disable something like sonar for example. Naze will not
fit everything.
Go through docs folder, there is a Section for developers how to setup
compiler