iNavFlight/inav #361: Why is custom motor mix not applied to basic throttle position requests?

DaleKramer, 2016-07-19:
It appears to me that custom motor mix is only applied to the calculated PID values in mixer.c here:

```
    // motors for non-servo mixes
    for (i = 0; i < motorCount; i++) {
        rpyMix[i] =
            axisPID[PITCH] * currentMixer[i].pitch +
            axisPID[ROLL] * currentMixer[i].roll +
            -mixerConfig->yaw_motor_direction * axisPID[YAW] * currentMixer[i].yaw;

        if (rpyMix[i] > rpyMixMax) rpyMixMax = rpyMix[i];
        if (rpyMix[i] < rpyMixMin) rpyMixMin = rpyMix[i];
    }
```

Why are they not applied to basic throttle position requests.  For instance, if CofG is positioned such that pitch mixer gives +1 strength to UP pitch motors and -0.8 strength for DOWN pitch motors, then there should never be a case where a throttle should request the same value for all motors (except maybe at min and max throttle and even that could be improved with some sort of motor mix factor).  To me it looks like every throttle request for all motors is equal and any motor mixing has to taken care of by the adjustment of the reactionary PID values.

Hard to explain my issue with this, but can someone see what I am asking and set me straight?

I know this gets into the fact that our thottle command of min to max throttle is not a linear relationship with actual thrust (which is another conceptual problem I have with PID values) but isn't the goal to have a 'best' guess at throttle values for requested stick position before you apply PID values?

Thanks,
Dale

digitalentity, 2016-07-19:
Throttle mix is applied later - https://github.com/iNavFlight/inav/blob/master/src/main/flight/mixer.c#L752

We calculate the RPY correction from PID output separate from throttle because it has no relation to overall thrust.

Throttle mix controls thrust response of propulsion system, while RPY mix controls angular rate response, the two are very loosly coupled.

DaleKramer, 2016-07-19:
I saw that but isn't the throttle parameter only used as follows with a value of 0 or 1?:

[image]

digitalentity, 2016-07-19:
The docs are probably outdated, nothing prevents  setting i.e. 0.8 as throttle mix, however you must be absolutely certain that motor don't contribute 100% of power to overall thrust

DaleKramer, 2016-07-19:
Interesting, then for sure the custom motor mixer calculator is wrong at [https://www.iforce2d.net/mixercalc/](url)

Also, once ROLL, YAW and PITCH parameters at set, is THROTTLE not just a calculation based on those 3?  Why even have THROTTLE parameter?

digitalentity, 2016-07-19:
You might have a pair of co-axial props in the center, spinning with the same RPM in the opposite directions - they are contributing only to thrust and have no effect on roll, pitch or yaw.

digitalentity, 2016-07-19:
Similarly, you may have a motor that contributes only to yaw, without affecting thust, roll or pitch (like a tail rotor in the heli)

DaleKramer, 2016-07-19:
Then my first question still remains,  with UP pitch motors set at +1 and DOWN pitch motors set at -0.8, then there is never a case where their basic throttle position before applying PIDs should be equal to each other and I don't see where in code that is done.

digitalentity, 2016-07-19:
If those motors are pointed straight down, they all contribute equal amount to overall thrust, however if they are at different distances from CoG they will contribute _different_ amount to pitch. I don't see the issue here.

DaleKramer, 2016-07-19:
But it appears that PID values are the only place pitch mixers are applied.  PID values are what I call reactionary values to obtain motor throttle commands that actually give you what you request with stick positions.

Don't you want to have your motor throttle positions at least as close as you can guess at them before applying PIDs?  In my example, the PIDs have to always make up for the fact that we always request equal throttle on all motors and in reality, a better guess would be to apply the pitch motor mix to the throttle before the 'motor mixed' PIDs.  Again, hard to explain.

digitalentity, 2016-07-19:
In case of a perfectly balanced aircraft and perfect propulsion system - you are right, but them we won't need PID controller at all. In reality pre-mixing the pitch will in fact de-stabilise the system as PID controller is unaware of that additional (feed-forward) throttle. 

In some cases a bypass for PID controller (feed-forward) is benificial - it increases speed of response, but it creates an open loop without gyroscope feedback. We then have two loops controlling the motor - closed loop though gyroscope and open loop through feed-forward. If they are properly balanced together they will increase the responsiveness of the system, however nothing is perfect and in general open-loop systems are less stable.

DaleKramer, 2016-07-19:
I know there is no feedback on this premix but it would be a closer guess to correct motor throttle position if it precisely represents the dynamics of the system and it would simply be like a constant that PID loop might be Ok with.  

I know it would further complicate debugging tuning issues if you describe the incorrect system with the custom motor mix.

In my mind it would make the job of the PID loop easier.

In any case, thanks,  I now understand why my EDIT2 comment of this post [http://www.rcgroups.com/forums/showpost.php?p=35260623&postcount=7190](url) did not make a noticeable difference to that issue.

closed 2016-09-01 by digitalentity