iNavFlight/inav #363: servos broken on 1.2 master

csurf, 2016-07-20:
downloaded & compiled 1.2 from master, configured flying wing, and servos are not functioning correctly on boot. The servos don't respond to RC commands, and twitch randomly to extreme positions. The solution is to connect to configurator, go into servos tab, and simply hit 'save'. Upon having the config re-saved, the servos begin working correctly.

My guess is that the servo config isn't being loaded correctly from EEPROM on boot. After booting up, the config looks ok on both the CLI and within servos tab in configurator, so the servo configuration *is* being stored correctly. However, for some reason, it doesn't load up properly, or there's an issue with the PWM setup that's causing invalid signals being sent to the servos.

digitalentity, 2016-07-21:
Which GCC version are you using?
There is a weird bug with servos. I was unable to locate it as it disappers when I try to debug it. I have a feeling that it is someshow related to link-time optimisations.

csurf, 2016-07-21:
using "gcc-arm-none-eabi-4_8-2014q2-20140609-win32" based on recommendation from CF wiki guide on how to build within Windows.

The strange thing is that it works as soon as I hit the 'save' button within the servos tab in configurator. Afterwards, the servos seem fine, but if I reboot the board, the servos break again, until I re-save the config. It seems as if there's a problem reading / loading the servo mixer config from EEPROM? What can I do to help troubleshoot?

digitalentity, 2016-07-21:
@csurf 
I believe this is a compiler/linker issue, because in my case simply adding `debug[0] = servo[0];` anywhere in the code instantly eliminates the issue. Try using a different version of the GCC, try 4.9.3, maybe try using git bisect to figure out the commit when it first happened. Thanks!

csurf, 2016-07-21:
@digitalentity,
ok, reccompiled w/ gcc  4.9.3 20150529, and the problem went away, so you were correct, it was a compiler/linker issue, which seems rather strange. Any ideas why this happened?

digitalentity, 2016-07-21:
@csurf No idea, I'm not a ARM assembler expert, hopefully somebody can look at the code generated by gcc 4.8 and see why its not working as expected.

sppnk, 2016-07-21:
I am building with gcc 5.3 and have no problems.

csurf, 2016-07-23:
I'll close this out once I get to do a flight test, but it appears to be fixed with the change in gcc.

stronnag, 2016-07-23:
I expect so, my tricopter servo has always worked with gcc 4.9.3 and 6.1 on iNav 1.2

closed 2016-07-26 by csurf