iNavFlight/inav #365: Extend available baud rates for serial ports

digitalentity, 2016-07-21:
Why are we limited to 9600-256000 for serial ports? I.e. LTM can operate at 2400 baud and MSP connection to PC can work up to 921600 baud.

This will, however break compatibility with Cleanflight Configurator.

jflyper, 2016-07-25:
For high bandwidth (not high speed) input support, we should review sizes of input bursts and their frequencies against processing bandwidth and  receive buffer sizes. For example, 921600 bps MSP connection, if any of the burst is longer than 256 bytes, requires taskHandleProcess() (and therefore cliProcess() or mspSerialProcess()) to be called at minimum of 360Hz with 256 byte receive buffer.

erpalma, 2016-07-28:
I think most of the traffic is directed TO the host, right?

BTW on F3 boards with VCP I think we might even go higher than 921600 (this is the limit for CP2102 used in many F1 boards). I'm not saying that we should, I'm just saying that it is technically possible.

jflyper, 2016-07-28:
@erpalma The direction doesn't matter. Since tx and rx operates at the same speed, if you want to output at 921600bps, then input comes in at maximum of 921600bps. If a head-to-head burst input exceeds serial task's processing bandwidth, then overflowed bytes are discarded:

> Serial task service frequency(Hz) * Receive buffer size (byte) * 10 (bits/byte) > Link speed (bps)

Such burst may be rare, even on the high/super/ultra-speed usb connections due to framing and transport protocol overhead, but we still have to be aware of it. This is what I meant by "Reviewing input burst sizes and their frequencies".

stronnag, 2018-04-04:
This was resolved some time ago, we have 1200 - 2500000 at least. Closing

closed 2018-04-04 by stronnag