iNavFlight/inav #371: In horizon mode, slow to auto level when release stick while exactly inverted

DaleKramer, 2016-07-23:
When I release my transmitter stick in horizon mode, while I am precisely inverted, I have had my craft take 8 seconds or more to return to a level attitude.  

When I release my stick in horizon mode a all other angles, my craft rights itself immediately (~1 second depending how far from level it was)

Is there a way to fix this or at least warn people that when flying precisely inverted, in horizon mode, your craft can be slow to recover when you release the stick (especially of interest to fixed wing craft)?

digitalentity, 2016-07-23:
@DaleKramer I assume you roll axis is still locked out.
The problem here is that "inverted" state is desribed as 180 deg *roll* not 180 deg pitch. Therefore, your machine will try to *roll* into upright position which is impossible to do as your roll axis is locked out.
This effect is due to we use euler angles to control the attitude.

DaleKramer, 2016-07-23:
Makes sense, yes still only pitch axis free to roam :)

Thanks

closed 2016-07-23 by DaleKramer