iNavFlight/inav #372: feature request: "Home Altitude" mode for RTH

csurf, 2016-07-23:
Implement a 'home altitude' mode similar to what EagleTree Vector has.

http://eagletreesystems.com/Manuals/vector.pdf (page 80)

this would require three new CLI parameters:
- 'home alt mode'
- 'home alt radius'
- 'home alt height'

'home alt mode' values:
- disabled: ignore 'home alt mode' settings
- normal: ascend / descend to 'home alt height' if within 'home alt radius'
- glide (only for FW); glide down with throttle off if within 'home alt radius' (perhaps we can improve on this to avoid stalls? maybe an option to glide in for landing with min throttle?)

The vector manual explains it pretty clearly. It doesn't seem like it should be too difficult to implement. Thoughts?

gaelj, 2016-07-26:
Sounds pretty nice to me.

digitalentity, 2016-07-27:
sounds like an interesting idea, but why bother? we are going to land anyway.

csurf, 2016-07-27:
home altitude can come in handy in certain cases. For example, in a situation where you lose FPV video and are far away from home and/or at high altitude, you can have the plane RTH at your pre-configured RTH height for the majority of the trip, but come down to a more manageable height at close range in order to make it easier to spot & maneuver the plane for a LOS landing (versus, for example, having the plane come home & perform a high-altitude loiter 300 meters up over your position).

Fast-Guy, 2018-05-20:
Where I fly it is very hilly and to clear the terrain I would have to set my RTH altitude to at least 2000'  to clear the hills. Once the plane flies home I might not be able to see it at 2000' to take over. This would be a great feature if it is possible. Thanks

fiam, 2018-06-24:
Home altitude part is done in #3431, the rest will be postponed to 2.1.

breadoven, 2022-11-26:
Isn't this effectively closed by https://github.com/iNavFlight/inav/pull/3431 ? Seems only the FW glide down is outstanding which doesn't sound particularly useful anyway.

closed 2023-10-27 by DzikuVx