iNavFlight/inav #378: WP-RTH - undefined behavior on a fixed-wing

digitalentity, 2016-07-23:
Landing procedure is not defined for a fixed-wing aircraft, normal RTH on a FW does not invoke landing, while WP-RTH will enter a multirotor landing procedure even on a fixed wing.

oleost, 2016-07-23:
Is it quick to get this defined? 

How about just simple loiter with target decent rate and no throttle?

oleost, 2016-07-23:
To get a target value I can try I do some flying in circles without motor and get the decent rate for my airplane.

digitalentity, 2016-07-23:
I think current landing procedure might work on a fixed wing with minimal
modifications. This needs to be further discussed and tested. Any input
will be greatly appreciated!

csurf, 2016-07-23:
some thoughts:
- user-configurable 'landing_loiter_radius' and 'descent_rate' params. 
- Would be cool to base descent rate on baro (if available) instead of GPS, which could cause some lag/lack of precision for this specific case. Possibly make it optional to enable baro for this? 
- completely killing the throttle too high might cause stalls & crashes on some airframes. Perhaps might be better to add a couple of params: 'landing_thr' & 'landing_glide_alt'. Use 'land_thr' for setting the throttle value to use while loitering&descending. Once 'land_glide_alt' is reached, then kill throttle & land in *straight, level* line (i.e., disable loiter). 

Ex: set 'land_thr' to something like 1200, and 'land_glide_alt' to 150-200cm in order to descend at low throttle, and glide in for landing from 1.5-2m up.

feedback welcome...

oleost, 2016-07-26:
@digitalentity Okey, Maybe I get to test it someday. To understand correctly failsafe RTH is still rth and loiter, while the new rth land waypoint is where its uncertain how the plane will behave.

I think PatrikE already have some sort of auto landing implemented, that we(ehm you) could have a look at. At the same try I try and contact him.

stronnag, 2016-07-26:
For your safety, mwp disables land on RTH when uploading missions for FW ...

digitalentity, 2016-11-17:
Fix pending by #799, closing.

closed 2016-11-17 by digitalentity