iNavFlight/inav #381: OSD support

digitalentity, 2016-07-24:
Support for boards with built-in OSD (SirinFPV)

digitalentity, 2016-07-24:
Blocked by #338

jflyper, 2016-09-28:
Anybody working on this?

DzikuVx, 2016-11-08:
@jflyper I've read on rcgroups you want to port OSD to INAV. have you started yet? I just got a board and I'm willing to proceed too

jflyper, 2016-11-09:
@DzikuVx Yes. Let's do this.
I think we can start with porting the MAX7456 driver (drivers/max7456.c); may have some trouble if bf and inav are not coherent with io management.
Upper layer (io/osd.c) has changed substantially to support Configuration Menu System (CMS), which was merged to master about an hour ago (see betaflight/betaflight#1422), and it's better to use this as a base.

DzikuVx, 2016-11-09:
Luckily both INAV and BF were moved to new IO so there should be no
problems.

jflyper, 2016-11-09:
Cool !!!

Things I've forgot to mention:
- PG migration status?
- Configurator

DzikuVx, 2016-11-09:
PG Migration is well... not there yet. If I remember correctly @martinbudden said few days ago that he will do it in BF and then in INAV.

I can handle Configurator

martinbudden, 2016-11-09:
PGs have been deferred to betaflight 3.2, and then to iNav. We'll eventually get them...

This port can be done with PGs though. And as @jflyper says, the new Configuration Management System (CMS) has just been merged into betaflight, and so that's a good place to start.

I think the port should be fairly straightforward.

One minor thing in betaflight "display" has been renamed to "dashboard" - that's a more accurate reflection of what it does, and it enables the term "display" to be used for anything that can display stuff, eg OLED or OSD (or indeed the display on the transmitter - there's work going on to get the CMS viewable on the transmitter screen - very useful for tuning in the field).

jflyper, 2016-11-09:
> I think the port should be fairly straightforward.

I think it's almost done ;)

https://github.com/jflyper/cleanflight/tree/inavdev-osd

- OSD on MAX7456 is working
- CMS on OSD is working
- OMNIBUS is the only target (for now)

@DzikuVx What board do you have?

digitalentity, 2016-11-09:
@jflyper I have an OMNIBUS. Do you want me to test something?

DzikuVx, 2016-11-09:
Omnibus F4 ... he he eh that was fast
I will add Omnibus F4 target to INAV soon and then I will test it there
too. And do Configurator

digitalentity, 2016-11-09:
@jflyper btw, send me an email (see my profile on Github), I'll send you an invitation to INAV Slack if you want.

martinbudden, 2016-11-09:
@jflyper , note that the iNav display (now called dashboard in betaflight) displays different information to the betaflight dashboard, and so should not simply be overwritten.

jflyper, 2016-11-09:
@digitalentity 
If you try it out, you will be the first one to see OSD and CMS run on iNav :) Will send you a mail.

@DzikuVx 
I'll be waiting for the OMNIBUSF4 target!

@martinbudden 
Yeap. I've noticed that. That's why it's not working right now. I will probably transplant the DISPLAYPORT on to the existing display.c (and may be rename it to dashboard.c).
MSP_DISPLAYPORT has some problem, too; it looks like the msp abstraction is one generation older compared to BF.

martinbudden, 2016-11-09:
@jflyper , I've just done a PR to rename display to dashboard.

I'll do another one to catch up MSP with betaflight - that should not be too difficult.

jflyper, 2016-11-09:
@martinbudden That would be great!

digitalentity, 2016-11-09:
@martinbudden thanks for PR, merge at will!

@jflyper, will try to test tomorrow. almost midnight here, he-he :)

DzikuVx, 2016-11-09:
@martinbudden @digitalentity before refactor MSP abstraction please merge #709 
I've done all I wanted in this branch code should be safe to merge to development since those are only new frames

digitalentity, 2016-11-09:
@DzikuVx merged

jflyper, 2016-11-09:
Okay, CMS over dashboard (formerly called display) is now functional.

@martinbudden Build option was bit different; DASHBOARD v.s. USE_DASHBOARD. Intentional?

martinbudden, 2016-11-09:
@jflyper , `DASHBOARD` was accidental. In my view all the build flags should be prefixed `USE_` - that way they are immediately identifiable as build flags. Leave it as `DASHBOARD` for the moment and I will fix it up later.

martinbudden, 2016-11-09:
There's a new PR https://github.com/iNavFlight/inav/pull/779 which moves over the betaflight MSP changes. I need to test that before merging it in.

jflyper, 2016-11-09:
@martinbudden 
I went ahead and cherry-picked dbbb66e out of #779.

The CMS is operational over MSP_DISPLAYPORT with SPRACINGF3 target now! :)

martinbudden, 2016-11-09:
OK, I'll move #779 in once the build passes, and then you can make your PR.

martinbudden, 2016-11-09:
@jflyper , I've got all my PRs in, you can now make a PR for CMS.

jflyper, 2016-11-09:
@martinbudden Thanks. Now the PR #782 is up.

DzikuVx, 2016-11-24:
OK, since we have implementation ticket in #817 I'm closing this one

closed 2016-11-24 by DzikuVx