iNavFlight/inav #390: Variable speed I2C and SPI

jflyper, 2016-07-25:
Not all devices are created equal. For SPI, some can talk at 15+MHz while others can only talk at 4MHz and slower. Similar on I2C; some can talk at 1MHz/800KHz, majority can talk at 400KHz, and rest can only talk at 100KHz. The situation I2C gets more complex because there are devices that can actually listen to faster clock, but can only talk at slower speeds.

It is nice to see if bus clock can be adjusted for a device that is being accessed. May be an additional parameter to i2cRead/Write/WriteBuffer().

This enhancement will broaden the design space for device placement on shared bus, and will probably be a great addition for the iNav-FC which explicitly supports multiple devices on the external SPI.

jflyper, 2016-07-25:
Is this already supported in sirinfpv code?

savaga, 2016-07-25:
No, it's not. The sirinfpv has all devices connected to different buses (3 SPI and one soft spi for vtx).

digitalentity, 2016-07-25:
Variable speed I2C is not possible - all devices are listening to the bus and processing data. If some device is unable to read data at some extreme speed it will likely cause corruption of data, received by that device, which, in turn, may cause the device to malfunction (treat received corrupted data as a register write or register read). Similarly for SPI.

closed 2016-07-25 by digitalentity

jflyper, 2016-07-25:
@digitalentity You are correct for I2C, but what about SPI? Authentic (correctly implemented) SPI devices are not listening to the bus unless selected, do they?