iNavFlight/inav #394: pos_hold_deadband influence the behavior of cruise mode

kronosol, 2016-07-25:
I´ve set pos_hold_deadband to 100cm.
I needed almost 25% of stick deflection to produce a reaction of the FC.
Maybe it´s because of the way cruise mode influence the PID controller?
At the moment i have no time, but i will try to upload a blackbox log till the end of the week.

digitalentity, 2016-07-26:
pos_hold_deadband is not in cm, it's in R/C command units and it's applied
to your sticks *after* the expo. You can set it as low as zero to have
instant response.

andre92, 2016-07-26:
shouldn't deadbands be applied before the expo?

DzikuVx, 2016-07-26:
According to the code, it is applied before expo
```
stickDeflection = constrain(rawData - masterConfig.rxConfig.midrc, -500, 500);
stickDeflection = applyDeadband(stickDeflection, deadband);
return rcLookup(stickDeflection, rate);
```

digitalentity, 2016-07-26:
@DzikuVx @kronosol 
R/C deadbands (deadband, yaw_deadband) are applied before expo.
NAV deadbands (alt_hold_deadband, pos_hold_deadband) are applied after expo in navigation code.

kronosol, 2016-07-27:
@digitalentity okay, but the wiki say´s it´s the deadband in cm.
[https://github.com/iNavFlight/inav/wiki/6.-iNav-CLI-variables](url)

> pos_hold_deadband	Input in [cm]	10	250
alt_hold_deadband	Defines the deadband of throttle during alt_hold [cm]	10	250

>

digitalentity, 2016-07-27:
@kronosol fixed, thanks.

closed 2016-07-27 by digitalentity