iNavFlight/inav #396: Mag Declination in pre-release 1.2 Master

Ralf-W, 2016-07-27:
Originally Posted by crashfrequently http://www.rcgroups.com/forums/showpost.php?p=35350224&postcount=7396

I have just compiled the latest 1.2 Master and found that adjusting Mag Declination when GPS is enabled has no effect on the heading displayed in the configurator. 
 If GPS is disabled, a new value entered and GPS is then enabled again the heading is adjusted by the new value.

theArchLadder, 2016-07-27:
Could this have something to do with the automatic mag dec feature? AFAIK it's enabled by default on F3 targets now...?

crashfrequently, 2016-07-28:
False alarm!
It was late at night when I was setting up the mag and I forgot that I was using a Dodo board with SPRACINGF3 code.

The reason for this is that I have not been able to compile working code for the Dodo from any 1.2 master after version 2f1ddedx dated June 14th. The compilation runs without error and I can flash the board but the validation fails with the error "STM32 - timed out".

When I posted the problem I was advised to use SPRACINGF3 code and that seems to work OK other than the reported mag declination oddity.

I think it's time to give up on the Dodo.

My thanks to all who responded to my post and my sincere apologies for wasting everyone's time.

Ralf-W, 2016-07-28:
I cannot find in iNAV, where the magnetic declination is transformed from "degrees,minutes" to deci-degrees. I do find it in Cleanflight code. 

In iNAV I do find in "imu.c"
77 float magneticDeclination = 0.0f;       // calculated at startup from config 

But where does this "calculated at startup from config" happen?

In "navigation_rewrite_pos_estimator.c" and in "navigation_rewrite_geo.c" I do not find any code for the case "NAV_AUTO_MAG_DECLINATION = FALSE".

digitalentity, 2016-08-02:
Mag declination is initialised from configurator at power-up or reboot currently. This is not bery user-friendly behavior and needs fixing.

digitalentity, 2016-08-18:
Not a critical bug, postponing for after 1.2

DzikuVx, 2018-06-05:
This thing is so old that we can close it

closed 2018-06-05 by DzikuVx