iNavFlight/inav #403: Limit rate acceleration

DzikuVx, 2016-07-28:
Idea by @borisbstyle already implemented in pidBetaflight: limit velocity/acceleration/deceleration inside rate controller. 

Might be great for big machines with huge inertia during stops. 
To consider: overshot might be better that oscillation 

See https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c#L278

digitalentity, 2016-07-29:
AFAIK it's not true angular accel limit, it's kI reduction if angular accel is above certain level. This might be a good idea, actually, acceleration is mostly handled by kP.

DzikuVx, 2016-07-30:
Let's say it's extension of what was ported lately from betaflight with yawItermIgnoreRate and rollPitchItermIgnoreRate. It not only limits Iterm growth when sticks are applied but only tries to limit acceleration. Boris states it's dps/ms^2 but I do not think so.
Anyway, this might improve things on heavy machines with huge inertia. It's better to overshoot and keep this overshoot in this case than overshoot and try to correct. Right now you have to learn not to release sticks too fast or it will oscillate

digitalentity, 2016-07-31:
I'm concerned with F1 targets. While they still have mostly sufficient performance, flash memory is at the limit. We will need to test if this accel limiting thing will really improve flight performance. I'm also thinking about making the PID controller code fixed-point or float point with an option to select one at compile time.

DzikuVx, 2016-07-31:
Honestly? I would drop F1 with 1.3. No new features for those

digitalentity, 2016-07-31:
I still want to support F1 boards, we'll need to make features selectable compile-time.

borisbstyle, 2016-07-31:
Or go with F4 :).

Keep adding math in there within issues.

Guys my "twitch" test for measuring motor response latencies works
incredibly well. Surprisingly very consistent.

digitalentity, 2016-07-31:
@borisbstyle, what's the "twitch" test? I'm a little bit out of the loop with Betaflight :smile:

borisbstyle, 2016-07-31:
Measuring and logging system delays in blackbox

See here an example of 8k oneshot42 on hqprop 5x4x3
https://www.dropbox.com/s/0bp7hl6tdglpoxl/oneshot42%208k%20HQ5x4x3.png?dl=0

It gives good understanding about system delays and it is really consistent.

F1 surely still works, but non being produced anymore. I dont see the
future in those tbh. I used to last year, but those are slowly dissapearing
from the market.

martinbudden, 2016-07-31:
I'm with @digitalentity here, I think we should continue to support F1 boards. They are still available and can be really cheap - I think it is important to support people who are on a budget (teenagers, for example).

I also like the discipline of having to produce 128K ROMs - it is a useful guard against code bloat.

And having boards with different processor speeds helps uncover timing bugs. Things might work "by luck" on faster processors, but there may be an underlying hidden bug that gets revealed running on an F1.

Although it may seem a pain at times, I think the discipline of supporting F1 boards results in better overall code quality.

(I'd actually go even further - I'd like to see F0 boards supported. I'd love to see betaflight in something like a CX10 or JJRC H20. See this - https://github.com/martinbudden/cleanflight/tree/CJMCU32K I got a sub-32K build of cleanflight, but configuration data blew the ROM budget by about 1K, and RAM was also an issue, but I do think this is possible)

digitalentity, 2016-07-31:
I actually have CX10, it would be interesting task to make it fly with INAV!

martinbudden, 2016-07-31:
The CX10 is a surprisingly good flyer, but the JJRC H20 hexacopter is even better. It's harder to fly, though and the yaw rates are crazy, but it is such fun. I'd love both to have rate mode though.

On of my long term projects that I am working to is a small (say 120mm or even smaller) iNav enabled brushed quad with GPS, optical flow, indoor positioning system, (something like this https://wiki.bitcraze.io/doc:lps:index), telemetry etc etc. This is one of the reasons I did the NRF24L01 code. It will probably take years to get all this in, but it will be fun.

digitalentity, 2016-07-31:
I'm thinking about buying a 3D printer and trying to build a <150mm GPS-enabled brushed quad (or Tri) myself.

martinbudden, 2016-07-31:
This guy does great frames for brushed quads. Lots of variety and also some of the frames have room to add extra stuff: http://picnicquads.com

I like brushed quads!

Linjieqiang, 2016-07-31:
@digitalentity The idea that  build a <150mm GPS-enabled brushed quad sounds great!

DzikuVx, 2016-08-16:
Since it's merged into 1.2 I'm closing it

closed 2016-08-16 by DzikuVx