iNavFlight/inav #412: Discussion - Convert PID math to fixed point

digitalentity, 2016-07-31:
A possible performance booster?

DzikuVx, 2016-07-31:
Yes, but not all. I would leave all Dterm connected parameters variables float, all Iterm connected fixed point where possible. And I have no opinion on Pterm

digitalentity, 2016-07-31:
Why keep D-term float point? Q16 have accuracy of ~0,00001, do we need more accuracy?

DzikuVx, 2016-07-31:
OK, I thought you wanted to go integer or smth like 2 decimal places. With 6 decimal places, I do not care

digitalentity, 2016-07-31:
No, I want to do a fixed point (libfixmath actually).

martinbudden, 2016-07-31:
I think it is definitely worth experimenting with this. I presume your plan is to have two versions of the PID, one fixed point and one floating point, but otherwise identical?

digitalentity, 2016-07-31:
@martinbudden one code, and a compile-time option to expand the code to float point or fixed point

martinbudden, 2016-07-31:
Yep, that works for me.

digitalentity, 2016-09-10:
I'm scheduling this for 1.3 - it would be a big space saver for F1 targets.