iNavFlight/inav #425: flip32 does not boot on c507aa2

stronnag, 2016-08-04:
as title, current master fails to boot on flip32 (blue light only). Not sure when this happened, as I mostly fly F3.

It also does not build with gcc 4.9, ok with 6.1

digitalentity, 2016-08-05:
Ok, I have a flip somewhere, I'll find it and find the exact commit that broke stuff

digitalentity, 2016-08-05:
Ref. baro: MS5611 was for some reason removed in affffb7ef1744bbe120baf56847f48cecb4d460c. @martinbudden can you shed some light why? :smile:

digitalentity, 2016-08-05:
Naze (Flip) boot issues introduced by 0489eb8b0840719277450a8d15bb6f1d5e6f62be

digitalentity, 2016-08-05:
Once I remove SONAR support from NAZE target it starts working again. Apparently SONAR support is broken in new IO

digitalentity, 2016-08-05:
More info - `#define SONAR` and `#define USE_RX_NRF24` can't co-exist. Once I undef either of them NAZE boots and works fine. @martinbudden can you have a look?

digitalentity, 2016-08-05:
No, it's deeper than that. Something is terribly wrong - board boots once in a while, the other time it hangs before detecting sensors. Possibly some race condition?

digitalentity, 2016-08-05:
Ok, I can't find a place where it can hang anywhere in the code. I need to hook up my JTAG and see where it's stuck.

digitalentity, 2016-08-05:
Ok, we are falling into `HardFault_Handler` from somewhere inside `SetSysClock`. Very interesting.

digitalentity, 2016-08-05:
HFSR = 0x40000000 (Indicates a forced hard fault, generated by escalation of a fault with configurable priority that cannot be handles, either because of priority or because it is disabled)
UFSR = 0x0001 (the processor has attempted to execute an undefined instruction)

How is this possible?

digitalentity, 2016-08-05:
Board now boots fine on each occasion. I'm even more confused.

jflyper, 2016-08-05:
Unexpected, sporadic interrupt diving into random location?

martinbudden, 2016-08-05:
@digitalentity MS5611 was removed as a ROM-saving measure. I erroneously assumed it was only used as an external barometer - I hadn't realised Flip32 used it. Sorry about that.

There is also en error in the current code in that   `EXTIInit();` is not called in`main.c` - this might be the root of this problem.

I have noticed that the order initialisation differs between iNav and betaflight - that may be problematic.

Can you try out https://github.com/iNavFlight/inav/pull/422 "New IO for UARTs, LED_STRIP and BIND_PIN " which has `EXTIInit` and see if this fixes the problem (I don't have a Flip board).

digitalentity, 2016-08-05:
I've put the `EXTIInit()` into `init()` but it didn't solve the problem. Board randomly crashes to HardFault from `SetSysClock()`.

martinbudden, 2016-08-05:
For F1 boards SetSysClock is defined as `void SetSysClock(bool overclock)` , probably not the problem, but odd that the compiler is not picking this up.

digitalentity, 2016-08-05:
Thanks for the hint. It appears that `SetSysClock` is called twice - from `init()` and from `systemInit()`, maybe this is the cause.

digitalentity, 2016-08-05:
F1 target clock initialisation tried to boot the board on HSI if HSE fails. I wonder if there is a target without an XTAL that actually requires this code or it can be safely dropped?

digitalentity, 2016-08-05:
IMO it was the doubled call to `SetSysClock()`, at least power-cycling the board 50 times didn't result in a single hangup.

martinbudden, 2016-08-05:
I love it when things can be debugged by reading the code.

digitalentity, 2016-08-05:
It took me two hours with debugger trying to figure out why SetSysClock() hardfaults, while I should have greped the source and checked for doubled calls...
I'm testing this on CC3D, NAZE and SPRacingF3, will merge after testing.

digitalentity, 2016-08-05:
Fixed

closed 2016-08-05 by digitalentity

stronnag, 2016-08-05:
@martinbudden you should upgrade the compiler, gcc 6.1 does catch the harmless declaration mismatch. 
https://github.com/iNavFlight/inav/pull/420#issuecomment-237671066

martinbudden, 2016-08-05:
@stronnag , you're right, thanks. Upgrading the compiler has been on my todo list for a while now...

PR #426 (2016-08-05): Fixes to initialisation procedure
1. Add missing call to `EXTIInit()`
2. Remove duplicated call to `SetSysClock()`
3. Remove duplicated FPU init on F3 targets
4. Re-enable MS5611 baro on NAZE target

Fixes #425