iNavFlight/inav #429: Before 1.2 release iNav should be checked for compatibility with betaflight/Cleanflight

martinbudden, 2016-08-06:
There will be differences, but we should check that all differences are deliberate rather than accidental.

digitalentity, 2016-08-06:
@martinbudden how do you suggest we do this?

martinbudden, 2016-08-07:
Carefully read the code! With special emphasis on the feature flags, CLI values and names and MSP values and names.

digitalentity, 2016-08-18:
@martinbudden can you attend to this? Your code reading skill is higher than mine :wink:

martinbudden, 2016-08-18:
I'll spend some time on this this evening.

martinbudden, 2016-08-18:
I've done some checking:

1. OK - Features enums `features_e`, defined in `config.h`
2. OK - MSP command `#define`s, defined in `map_protocol.h`
3. MSP serialisation. Some differences, but OK because of the different features implemented in iNav and betaflight.
4. CLI commands.  Some differences, but OK because of the different features implemented in iNav and betaflight.

There were some differences between the `boxes` in `serial_msp.c` e.g.:
**betaflight**
```
    { BOXAIRMODE, "AIR MODE;", 28 },
    { BOX3DDISABLESWITCH, "DISABLE 3D SWITCH;", 29},
    { BOXFPVANGLEMIX, "FPV ANGLE MIX;", 30},
    { CHECKBOX_ITEM_COUNT, NULL, 0xFF }
```
**iNav**
```
    { BOXNAVWP, "NAV WP;", 28 },
    { BOXAIRMODE, "AIR MODE;", 29 },
    { BOXHOMERESET, "HOME RESET;", 30 },
    { BOXGCSNAV, "GCS NAV;", 31 },
    { BOXHEADINGLOCK, "HEADING LOCK;", 32 },
    { BOXSURFACE, "SURFACE;", 33 },
    { BOXFLAPERON, "FLAPERON;", 34 },
    { BOXTURNASSIST, "TURN ASSIST;", 35 },
    { CHECKBOX_ITEM_COUNT, NULL, 0xFF }
```

I'm not sure if we should align these box values.

digitalentity, 2016-08-19:
Good, thanks!

Regarding boxes - I don't think we should align boxes, at least not at the moment. In future - maybe, especially when PGs will be introduced as then some of the configuration might survive the firmware change and it would be wise to keep those parts of the configuration compatible.

digitalentity, 2016-08-19:
@martinbudden  if you think that we've checked everything we need - you can close this issue.

closed 2016-08-19 by martinbudden