iNavFlight/inav #431: RFC: Do we PH random failures post 43eaf10 ?

stronnag, 2016-08-06:
Today I flew 85fe245 and f440ead on my usually utterly reliable quad. This machine has not failed to execute PH, RTH or missions since we fixed SBAS months ago ---- until today.

Result. Completely random PH behaviour (if PH fails, then I don't even try any other nav function).  Randomly, on hard reset, nav functions will either work or fail. Some log files at 
 http://seyrsnys.myzen.co.uk/inav_ph_woes/. All the log files were created in a short time frame with good satellite coverage (16-19) and good HDOP (1.1 - 1.3).

- LOG00279.TXT: 85fe245. PH fails miserably, repeated.
- LOG00280.TXT: f440ead (having persistent mag/acc makes checkout / build / flash / apply config in the field so easy:). Works OK.

... does some betaflight stuff on the minis for 30 minutes

- LOG00281.TXT: f440ead: nav functions fail miserably, repeatedly (multiple arm / disarms)
- LOG00282.TXT: f440ead: *powercycle* nav functions work again !!!!
- LOG00283.TXT: f440ead: *powercycle* nav functions still work. Out of lipos, otherwise I'd have tried 43eaf10 (which behaved perfectly across multiple lipos earlier in the week).

**Exec Summary** : nav post 43eaf10 seems pretty random to me at the moment. Other experiences solicited.

digitalentity, 2016-08-06:
Seems like a GPS-related problem. GPS (and estimated navVel) velocity doesn't seem to be correlated with UAV attitude and stick input:
[image]

It might also be a compass issue:
At log 281 there is a moment when UAV is flying at only forward pitch - this is usually flying straight forward and GPS heading should be roughly equal to magnetic heading, however they are opposite:
[image]
[image]

However, I doubt it's compass - UAV heading strongly correlates with yaw stick input. I'd blame GPS reception that reported invalid coordinates/velocities/course for some reason.

@stronnag was there strong wind that day, if so - what direction?

stronnag, 2016-08-07:
@digitalentity, thanks for the analysis.

There was less than 8 m/s wind; I often fly in more. My initial thought (from the field) was that it was just a bad GPS day. I guess that happens from time to time. I did not suspect the compass, much of random flying / drifting around was manually correlating the observed heading with the LTM reported heading (which seemed consistent), whereas the mwp calculated 'range from home' often seemed incongruous. 

It was an object lesson in 'try PH before engaging more advanced nav functions'.

I'll do some more missions today before closing.

digitalentity, 2016-08-07:
@stronnag 
Interestingly GPS glitch detection did fire an alarm a few times. We definitely need better glitch detection logic.

stronnag, 2016-08-07:
A day later .... ipdated the firmware to abf1015 (so no significant change). Starts with similarly less than stellar sat statistics (13-15 sats, 1.3-1.5 HDOP); couple of hours later its back to normal, 19-20 sats and 1.1 HDOP.

PH and other nav functions back to their awesome performance and consistency, even at the lower than normal end of the sat coverage range. Someone in the DoD really didn't like me yesterday.

Maybe we should use some of those spare X-FRAME bits for a glitch detection alert or counter?

digitalentity, 2016-08-08:
@stronnag thanks for your report, indeed something must be wrong with GPS reception. We can use a byte in X-Frame to indicate INAV internal status flags, however we still don't have a good logic of detecting GPS failures...

Linjieqiang, 2016-08-09:
The solution to the problem is using EKF algorithm although it looks messy and complex.

digitalentity, 2016-08-09:
EKF is nothing close to a glitch detection and protection, it's merely an algorithm to blend data from available sensors.

Linjieqiang, 2016-08-09:
If large difference between receiving from gps and predicting form EKF,program will move the data from gps.Maybe I'm wrong.

digitalentity, 2016-08-09:
It's not the EKF itself, it's a supporting code logic that does the detection.

Linjieqiang, 2016-08-09:
Oh.Sorry.It's my wrong.

xdigyx, 2016-08-10:
Today I had a situation when at pos hold mode my hexacopter rapidly started to fly away. Not sure whether it was caused by loosing some sats or... Anyway maybe it would be a good idea either to take an average of 2-3 readings or ignore for pos hold mode position change grater than... (say 1m/sample)?

stronnag, 2016-08-10:
GPS glitches are very rare, my report was a once a year occurrence, for a frequent flyer.  Your problem is more likely symptomatic of mag interference. Have you verified that the mag works perfectly at all throttle settings?

xdigyx, 2016-08-11:
Not sure what's telling you it's mag problem. If it was, then I believe the model would make slowly bigger and bigger circles, but it would not rapidly fly away. Do you agree?
Although I have used pos hold max for 1h, it happened only once, apart this time pos hold works great.

digitalentity, 2016-08-11:
Hard to diagnose w/o logs. Circles happen when heading is slightly wrong (up to maybe 30 deg) - this result in quad moving in **slightly** wrong direction. Bigger error will result in quad going in significantly wrong direction and a case of fly away.

xdigyx, 2016-08-11:
There were no circles, just pitch inclination increase by approx 30 deg. Now I know how to ebable onboard dataflash bbox, so hope to catch it next time.

digitalentity, 2016-08-11:
It's either a magnetic anomaly or compass failure, or real GPS glitch. If compass wiring is not good the sporadic bug in connectivity to compass chip may cause the chip to freeze and give out same heading over and over again. Please also check the wiring to the compass.

digitalentity, 2016-08-11:
#453

xdigyx, 2016-08-11:
I am using onboard compass, so no wiring, can try to threat it with hot air but low chance. The FC is shielded (50x50mm 35um Cu PCB) from the bottom and grounded.

DzikuVx, 2016-08-11:
@xdigyx shielding like that would not block magnetic field from power cables. It might shield oscillating magnetic field, but frequency would have to be > 100kHz. 
Interference from power cables is almost not oscillating, grounded Cu PCB or even full Faraday cage is ineffective. 
The only solution is to move power cables from battery/ESC further away from FC and (better option) use external magnetometer on a mast

xdigyx, 2016-08-11:
Got your point. Firstly I will try to catch the event with blackbox to see what's the root cause and maybe discuss what can be done to avoid such situation . Then I will use an external amg.

stronnag, 2016-08-12:
@digitalentity. I no longer think that this is an external GPS issue. 

* I've been flying this quad for months without nav issue; I have 100s of logs to prove it;
* The problems *coincidently* only happened after major I/O changes late in the 1.2 cycle.

Today. 
* Arm, PH fails miserable (no log alas, straight line 'fly away').
* Land. **Power cycle.** PH works LOG00292.TXT /1.
* Land. Disarm (still powered), fly a mission perfectly.  LOG00292.TXT /2
* Land. **Power cycle.** PH Fails miserably (straight line 'fly away') LOG00293.TXT
* Land. **Power cycle.** PH works perfectly. LOG00294.TXT

(all logs at http://seyrsnys.myzen.co.uk/inav_ph_woes/). There was perhaps 30 seconds in the land / power-cycle sequence. I find it hard to believe the satellite performance is varying in that short time period. The whole sequence described above was within 7 minutes, with consistently between 17-19 satellites and c. 1.1 HDOP. 

That I can have nav functions randomly work / fail on power cycle within very short time periods looks to me like a firmware issue rather than a celestial GPS issue.  I'm encouraged in this theory by your recent CC3D sensor woes.

martinbudden, 2016-08-12:
I'm certainly willing to believe this is a firmware problem. The IO changes were pervasive, and although I was very careful, there is certainly a possibility that I've introduced a bug somewhere.

stronnag, 2016-08-12:
It's all circumstantial at the moment, but definitely a regression since 2016-08-04, the last flawless nav experience.

xdigyx, 2016-08-12:
Today I did catch same issue. For sure there was some yaw drift, but it seems like the model was all time trying to face the starting position. After approx 2 min from start num of sats dropped to 0 just for one read cycle and my model rapidly moved.
I've got the log file. Some lines here:
time (us)	 GPS_fixType	 GPS_numSat	 GPS_altitude	 GPS_speed (m/s)	 GPS_ground_course	 GPS_hdop	 GPS_eph	 GPS_epv
105125924	2	16	123	 0.06	 181.4	129	88	132
105326384	2	16	123	 0.06	 181.4	129	88	132
105526783	0	0	123	 0.07	 181.4	9999	95	132
105727183	2	16	124	 0.07	 181.4	129	95	132
105927635	2	16	124	 17.76	 171.4	131	95	132
106128086	2	16	124	 0.56	 171.3	131	95	132
106328488	2	16	124	 0.11	 171.3	131	95	132
106528932	2	16	124	 0.31	 127.9	131	94	132
106729408	2	16	124	 0.59	 22.1	148	94	132
106929832	2	16	124	 0.86	 359.3	148	94	132
107130275	2	16	124	 0.86	 359.3	116	94	132
107330739	2	16	124	 0.83	 356.9	122	94	132
107531103	2	16	124	 0.82	 357.4	119	94	131
107535112	2	16	124	 0.82	 357.4	119	94	131
107731575	2	16	124	 0.65	 355.4	119	94	131
107931951	2	16	124	 0.44	 355.3	122	93	131
108132427	2	16	124	 0.34	 357.9	122	93	130
108332876	2	16	124	 0.20	 0.2	129	93	130

The full log I can send on email to whom it may concern (pm me).

digitalentity, 2016-08-12:
@stronnag, I agree, this is very likely software issue. However I'm
thinking it was there before the IO changes - I remember having odd mixer
issues a while ago **before** that major change happened.

Which direction was your machine facing on powering up? Maybe something in
firmware is messing up the mag...

digitalentity, 2016-08-12:
This is going to be a tough one - changes to firmware will affect the
memory layout and may seemingly "fix" the bug...

digitalentity, 2016-08-12:
@stronnag I have a strong feeling that something is wrong with either IMU of magnetometer code.

From your latest logs:
**LOG 292**
heading 31 - PH works

**LOG 293**
heading 22 - PH fails
heading 28 - PH fails

**LOG 294**
heading 200 - PH works

Interestingly in LOG292 and LOG293 machine does the same (correct) tilt to corrent for error however actual correction differes. This can only happen when heading is incorrect.

Can you send exactly the same hex/dump file you've been using during these tests so I can check it on my SPRF3 board?

digitalentity, 2016-08-12:
@xdigyx you likely experienced an ordinary GPS glitch, might be GPS wiring as well.

digitalentity, 2016-08-13:
I have a feeling that recent CC3D woes and this issue are related...

xdigyx, 2016-08-13:
Yes, I agree. Checked the wiring and no loosen wires or connectors. Anyway, glitch was not detected and caused model to fly away, so just a proof that better glitch detection is still needed.
Thanks

stronnag, 2016-08-13:
@digitalentity. I tend to do my PH tests with the craft 'beam on', so c. N / S heading is expected. If failed earlier on c. 200°, and I've had failures on other orientations. I (almost) always power up at 110° - 120° (facing away from 'my' bench), which is my first mag pre-flight check. 

The hex (gcc 6.1 compiled) is at http://seyrsnys.myzen.co.uk/inav_ph_woes/inav_1.2.0_SPRACINGF3_8cfc74b.hex

From my records (I keep all my logs), it seems https://github.com/iNavFlight/inav/commit/43eaf10db2170633cf424baa71a6bf0082c8061f was the last non-affected build; I will fly that today.

oleost, 2016-08-13:
Maybe an announcement on forum regarding this?

stronnag, 2016-08-13:
Yes, I will do one immediately.

digitalentity, 2016-08-13:
@stronnag, @oleost yes, thats a good idea, thanks!

@stronnag can you also provide a dump so I'll be testing exactly the same fimrware setup?

stronnag, 2016-08-13:
cli dump http://seyrsnys.myzen.co.uk/inav_ph_woes/nav_sprf3_fp.txt

digitalentity, 2016-08-13:
@stronnag thanks. Now I should try to find a GPS module...

digitalentity, 2016-08-13:
> From my records (I keep all my logs), it seems 43eaf10 was the last non-affected build; I will fly that today.

@stronnag it would be awesome if you could pinpoint the commit that started to glitch. I suspect that this particular issue might be related to a buffer overflow or maybe a race condition in some interrupt handler since it's intermittent. I suspect UART - if an interrupt handler somehow gets invoked before buffers are properly initialised it might result in overwriting arbitrary memory (please correct me if i'm wrong).

stronnag, 2016-08-13:
I'm going out this after noon armed with all my big quad LIPOs and 
````
inav_1.2.0_SPRACINGF3_43eaf10.hex
inav_1.2.0_SPRACINGF3_4bbd176.hex
inav_1.2.0_SPRACINGF3_8cfc74b.hex
inav_1.2.0_SPRACINGF3_f440ead.hex
````
I expect 43eaf10 to work and f440ead not to work, based on previous experience.

The great thing about having mag and acc saved in the dump file is that reflashing / return to known config in the field is so, so easy.

digitalentity, 2016-08-13:
@stronnag thank you so much for testing this out! I'm struggling to figure out what's going on with CC3D - it's very repeatable glitch as I have a feeling it's related to GPS issues.

stronnag, 2016-08-13:
27822a5 is the last reliable commit. 6+ power cycle / PH successful attempts. All later builds will fail PH with 2 attempts. more details & logs when I get home.

stronnag, 2016-08-13:
Here's the full bisect test results.
**Methodology:**

1.  stm32flash the required firmware
2. use mwptools cf-cli tool to install known configuration
3. Perform PH
4. Land (auto if PH works)
5. Powercycle
6. repeat step 3 until PH FAIL or 6 consecuitve PASS

**The results being:**

| Commit | Date | Result |
| ------------- | -------- | ---------- |
| 43eaf10 | 2016-08-02 | **FAIL** : LOG0295.TXT |
| 742f429 | 2016-07-31 | PASS : LOG0304.TXT 
| 742f429 | | **FAIL** : LOG0305.TXT |
| 0489eb8 | 2016-07-31 | PASS : LOG0296.TXT |
| 0489eb8 | | **FAIL** : LOG0297.TXT |
| 27822a5 | 2016-07-29 | PASS : LOG0298.TXT |
| 27822a5 |  | PASS : LOG0299.TXT  | 
| 27822a5 |  | PASS : LOG0300.TXT  |
| 27822a5 |  | PASS : LOG0301.TXT |
| 27822a5 |  | PASS : LOG0302.TXT |
| 27822a5 |  | **PASS** : LOG0303.TXT |

Several successful WP and RTH mission were subsequently flown with 27822a5.

All log files under  http://seyrsnys.myzen.co.uk/inav_ph_woes/

I trust this is conclusive. It also supports the tentative conclusions relating this fault to the CC3D fault and the new IO subsystem.

xdigyx, 2016-08-13:
**stronnag**, I have just compared my dump and yrs and noted one strange value:
set mag_declination = -130, should it not be positive approx 110 POSITIVE?

stronnag, 2016-08-13:
Not if (a) one habitually flies in the New Forest area of southern England, where the declination is c. 1° 30' W and (b) you have `set inav_auto_mag_decl = ON` set.

May I also suggest that this thread is for trying to fix a serious regression in iNav, not provide basic support to beginners; please use the RC Groups topic <http://www.rcgroups.com/forums/showthread.php?t=2495732> for basic support.

digitalentity, 2016-08-14:
@stronnag thanks for narrowing this down. There still is a possibility that in 27822a5 and earlier this bug is still present - there was a bug reported with frozen servos on SPRF3 which I also suspect to be a result of memory corruption. Coincidentally that bug also happened after beginning migration to new IO.

@martinbudden, @ledvinap if you have the time please have a look at the code to see if there's anything wrong with new IO code review in INAV.

digitalentity, 2016-08-14:
We are lacking any protection from wrong indexes, incorrect pointers etc. I think we should start putting `ASSERT(x)` all over the code with three possible compile-time options:

1. ignore (tested production code)
2. record a line to a given location in memory
3. same as 2 but with additional hardfault

Thoughs?

digitalentity, 2016-08-14:
For one:
```
ioRec_t* IO_Rec(IO_t io)
{
    return io;
}
```
this assumes `io` is correct but what if it's not? then pointer to non-existent `ioRec` will be returned with a possibility of writing `owner`, `resource` and `index` to arbitrary memory localtion causing corruption of data.

xdigyx, 2016-08-14:
Do not be so rude, I just wanted to help. Sometimes even basic things can be overlooked by pro users... 110 or 130, but I believe it should be positive (unless FW does not look at it at all and it really does not matter).

digitalentity, 2016-08-14:
@xdigyx I think @stronnag didn't ment to be rude, mag declination indeed varies greatly between different places. However this issue is related to latest 1.2 code and tests on 1.1 are probably not relevant, but they create noise that destract attention from the topic.

sambas, 2016-08-14:
Tested current master (8cfc74b) with revo. Didn't have any notisable issues. 5 short flights (PH spin+ RTH) with power cycle between. Only three fitted in the bb log. Build with gcc-arm-none-eabi-4_9-2015q2-20150609-win32 using command "make -j24 TARGET=REVO OPBL=yes". Included map file if you could provide same with gcc 6 to compare.
https://www.dropbox.com/sh/tucne4xbeanyqax/AABw0wqCcqdb0lyaBVCyD7Aha?dl=0

digitalentity, 2016-08-14:
I don't have any GPS-related issues on my Y6 built with GCC-4.9.3 as well, although CC3D won't work at all...

stronnag, 2016-08-14:
Strangely, I don't (think I) have this issue on my Dodo either --- I will test it rigorously later today (the problem manifests itself on an acro SPRF3 with external I2C BMP280 + M8N (I2C compass)).

@sambas : I've uploaded gcc 6.1 maps for SPRF3, REVO, NAZE and CC3D to http://seyrsnys.myzen.co.uk/inav_ph_woes/

xdigyx, 2016-08-14:
Comming back to the topic of GPS glitch...

In my case there were a difference in position of 4.5m between one and following GPS pos reading after a glitch, but this was not caught.
I am not an expert, but just a simply thought:
At the moment prediction of location is a radius of a constant value (INAV_GPS_GLITCH_RADIUS + 0.5f * INAV_GPS_GLITCH_ACCEL * dT * dT) this probably is enough for nav, but for PH the value is quite huge. 
Maybe "last known velocity"*dT*c ("c" = a constant in place of INAV_GPS_GLITCH_RADIUS, say 1.2 or so)   would be a better estimation?

martinbudden, 2016-08-14:
Looking at the work @stronnag has done:
https://github.com/iNavFlight/inav/commit/27822a59cb849266427084508146fab425d6a037 was the last reliable commit. The next significant commit after that was https://github.com/iNavFlight/inav/commit/0489eb8b0840719277450a8d15bb6f1d5e6f62be - Converted I2C, EXTI and SONAR to new IO, which didn't work.

However there was a problem with that commit in that I forgot to add `EXTIInit` in `main.c`, which I corrected in a subsequent PR. So, unfortunately narrowing it down to that PR does not really help us - it didn't work because of an unrelated issue.

@digitalentity , I think there is some merit in adding asserts, but I think that it's unlikely the problem is in the driver code, since no-one has reported a similar bug in betaflight. On the other hand, perhaps no-one has built betaflight using 4.9.3. Can you try building betaflight with 4.9.3 and see if the CC3D issue occurs - if it does then we know there is a problem with the underlying code, rather than the port.

The device drivers are virtually the same between betaflight and iNav, so my thoughts are that it could be a problem with initialisation or in the sensor code.

Also, can you try building PR https://github.com/iNavFlight/inav/pull/456 with 4.9.3 and seeing if the problem exists there - this PR is further convergence of the device drivers between iNav and betaflight.

digitalentity, 2016-08-14:
@martinbudden the problem is that the CC3D bug goes away even if I add `debug[0] = gyroADC[0]` line in the gyro.c, so that PR seemingly might fix the issue without actually doing so.

martinbudden, 2016-08-14:
Yes, you're right. We can't say this bug is fixed until we actually understand what the problem is.

stronnag, 2016-08-14:
@martinbudden. No build after https://github.com/iNavFlight/inav/commit/27822a59cb849266427084508146fab425d6a037 works consistently on my previously super-reliable quad (SPRF3 / BMP280/ M8N), including builds that fixed `EXTIInit`. The failing logs from Friday https://github.com/iNavFlight/inav/issues/431#issuecomment-239489749 were from 8cfc74b (current master).

However, it's deeper than that. Today I tested the F3/Dodo/M8N/TRI on 8cfc74b;  of course that works perfectly, including my arbitrary "6 strikes till you're out'" power cycle / PH test.

To my simple mind,it appears  something post 27822a5 causes random initialisation failures on some hardware. Not a lot of help in narrowing the problem down. I appreciate there have been no similar reports on Betaflight; this does not really surprise me, as the use cases (and use of sensors) are quite different. I fly βF a lot; just not with many sensors enabled.

stronnag, 2016-08-14:
It appears to me that the new IO subsystem is less tolerant of I2C daisy chaining than the previous implementation

| Configuration | Average failure of PH post powercycle |
| ----  | ---- |
| Dodo FC. BMP280 on FC,  HMC5883 on GPS via I2C | No failures |
| SPRF3 Acro. BMP180 + HMC5883 on external board, via I2C  |  1 in 4 attempts fails |
| SPRF3 Acro. BMP280 on I2C, daisy chained to HMC5883 on GPS | 1 in 2 attempts fails |

i.e. the I2C with two daisy chained items is the worst performer with 8cfc74b; all configurations perform equally well (no PH failures) with  27822a5.

The latter two examples are the same machine, reconfigured. The BMP280 configuration (worst case) was the subject off the previous tests.

digitalentity, 2016-08-15:
@stronnag interesting finding. I now remember solving I2C overclock issues when doing some I2C-related changes in INAV quite a while ago. Since now we have I2C drivers from Betaflight we might be hitting the overclock issue again.

The fact that problem occurs more frequently with longer wires only confirms the thory.

I'll hook up my logic analyser to SPRF3 and verify today.

digitalentity, 2016-08-15:
I think I've found something.

With new IO drivers SCL and SDA pins are initialised as IOCFG_AF_OD. On both F1 and F3 boards this result in GPIO port output speed register being set to zero (2MHz clock). On longer lines this may cause unacceptable rise/fall times as seen by CPU which in turn may cause data corruption. 
I'm not sure how this affect sensor readings and why I didn't notice anything significant in the logs, but prioir to new IO I2C pins were clocked at 50MHz.

#465 intends to fix the issue.

digitalentity, 2016-08-15:
@stronnag can you verify the latest master on your problematic setup?

[126 more replies]

PR #517 (2016-08-29): Add extra delays when cold-booting the UAV
This is an attempt to exploit @stronnag's finding in https://github.com/iNavFlight/inav/issues/431#issuecomment-242983046

Fixes #431
closed 2016-08-29 by digitalentity