iNavFlight/inav #444: Board Alignment

taileron, 2016-08-09:
... wanted to test inav in my projects (tilt multis with foam frame) the fc has to be mounted (glued) up side down it seems that the settings for example roll 180 degrees doesn´t apply.
How can I activate MW 2.3 pid the only one with enough yaw authority in my case. It switches back to Luxfloat every time. 
For a tilt tricopter I need more than one smix servo.
(currently it runs cleanflight 1.13 on SpRacingF3)

digitalentity, 2016-08-09:
INAV has only one PID controller. It's neigher Luxfloat not MW23.
You need to increase yaw_p_limit, increase yaw rates to your liking, and tune yaw P.
For board alignment to work either use CLI or INAV Configurator. INAV board alignment is in deg*10 so you'll have to use 1800 = 180deg
Please also consider reading the [wiki ](https://github.com/iNavFlight/inav/wiki) before opening issues.

closed 2016-08-09 by digitalentity

taileron, 2016-08-15:
…. alignment works perfect now, thanks and I like the flight-behavior of Luxfloat much more than MW 23 in most situations.
I´ll give it a try in this project as well
Tried now 1.12 RC1 
Now custom and custom tri support also two servos for tilt systems that´s great for me …..

But as soon as there are more than 3 mmix entries (in custom or custom tri mixer)
yaw_p_limit (cli set to 500) is limited to 200 on all servo outputs used by smix with stabilized yaw as source